ArduPilot/ardupilot #10565: Larger ANGLE_MAX considering normal wind speed.

p2041, 2019-02-19:
### Bug report

**Issue details**
Hello.

I think ANGLE_MAX's documented ranges should be modified. http://ardupilot.org/copter/docs/parameters.html
The current range is from 1000 to 8000.
However, if there is some wind (e.g., 10m/s), a drone may be drifted to the different direction in SITL. (quadcopter with IRIS+ model).
10m/s is a quite normal wind. So, I think it would be great if a drone resists such wind. 
So..... probably 2000 looks better.

Also, I found that ANGLE_MAX can have 0, which causes a drone to be stuck in the sky.
I think it would be better to limit its ranges.

**Version**
ArduCopter 3.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[O] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadcopter

**Hardware type**
SITL

rmackay9, 2019-02-19:
So the parameter range is from 1000 to 8000 centi-degrees which means 10degrees to 80degrees which is quite a wide range.  The default is 4500 (45deg) which is perhaps too high but unrelated to this issue I think.

I agree we should not allow ANGLE_MAX to be zero so I'll change the title of this issue I think.

ktskt, 2019-02-19:
@rmackay9 Oh, my description was wrong. My suggestion was that (although my writing was very confusing) I think the range of ANGLE_MAX should be from *2000* to 8000 --- not from 1000 to 8000. 
So... what I intended was 1000 should be larger.

As I described, with 10m/s, a drone may be drifted with 1000 centidegree (due to limited pitch angle). If it is 2000, it was fine.
I think 10m/s is normal wind although it is a little bit strong. 
That's why I suggest changing the minimum value of ANGLE_MAX to 2000.

p2041, 2019-02-19:
Sorry, it is wrong description. My suggestion is that I think the range should be from *2000* to 8000. (1000-> 2000 means larger range).
If there is some wind (e.g., 10m/s), a drone may be drifted since the pitch angle is limited to low angle, which cannot generate enough force to resist wind.

Perhaps, a drone with a powerful motor could solve that problem. But, that may not be deployed on every drone I think.

rmackay9, 2019-02-19:
@p2041, no troubles.  I see what you're saying but we don't really know special situation the vehicle might be in.  10 deg is not enough outdoors but it could be enough for indoors so I think we can leave the lower limit as 1000..

p2041, 2019-03-21:
@rmackay9 I thought many drones may fly outside. But... I agree that 1000 is ok for indoor cases.
If you think it is too high value, it is fine. 
But, please note that, in the outside, 10m/s is somewhat strong but normal wind according to the wind statistics.

peterbarker, 2019-03-25:
Wouldn't an ANGLE_MAX of zero make sense for some frame types?

rmackay9, 2019-03-25:
@peterbarker, not for any of the frames that we currently support.. we can imagine frames that might want and angle-max of zero but we can address that if such a frame ever appears..

IamPete1, 2021-07-26:
We do have a arming check on angle max. We also never use a angle of less than 10.
https://github.com/ArduPilot/ardupilot/blob/f20644b6cf87b53c2394ad1ff07673c5ae657864/ArduCopter/mode.cpp#L385

I'm not really sure if there is anything more we should be doing?

rmackay9, 2021-07-27:
I agree that I think this issue is covered so let's close it.

closed 2021-07-27 by rmackay9