ArduPilot/ardupilot #10566: Prearm warning board voltage out of range on pixhack v5

bnsgeyer, 2019-02-19:
### Bug report
compiled latest master  (f8e49e5) to target "fmuv5" and flashed to a pixhack v5.  After rebooting the controller, a prearm message appeared.

Prearm: Board Voltage (0.0v) out of range...

**Version**
 master 3.7dev (f8e49e5) 

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
tradheli
**Hardware type**
pixhack v5
**Logs**
No log.  I can provide one if it is necessary.

WickedShell, 2019-02-19:
Arming check came in here: https://github.com/ArduPilot/ardupilot/pull/9714

Unless I've missed something we were actually checking the board voltage before since FMUv5 has the `HAL_HAVE_BOARD_VOLTAGE` defined. I'd look through your logs to check if you ever had working board voltage monitoring.

bnsgeyer, 2019-02-26:
@WickedShell I ran a log file and ran it disarmed on flight battery power.  I am still getting the prearm warning that doesn't allow me to arm.   So I am not sure that whether this is a software or hardware thing.  I am running this on a developer version of CUAV pixhack v5.  So I have tried it on both power ports and get the same result.

[log file](https://drive.google.com/open?id=1ysCh3Vw7Gx_tlejGuYESizaQX9fFB9Rm)

bnsgeyer, 2019-02-26:
@ChristopherOlson have you loaded a 3.7 firmware and tried to arm on flight battery power?  I tried it tonight and even on battery power I got the prearm warning.  Just trying to get someone who can verify that its not just me.

WickedShell, 2019-02-26:
@bnsgeyer Your logs indicate that Vcc is impossibly low. (POWR.Vcc). Either the hwdef has the wrong pin, the generator has a problem, or there is a problem with the board. We only run the check if the hwdef reports that there is a pin for Vcc monitoring.

ChristopherOlson, 2019-02-26:
@bnsgeyer no, I have never tried it with 3.7. Did you try inserting a thin probe in the plug and measure the actual voltage from the power module with the system powered? Did it work ok on 3.6 and the problem is only on 3.7?

With 12S power system it is usually recommended to use either one power module on a 6S side, or use dual power modules (if you wish to monitor both 6S batteries) on both 6S sides. Even though they claim the PM will handle 60V input, that is kind of "iffy" stepping 45-50V down to 5.2V on a helicopter that can easily pull 120A if you lean on it.

bnsgeyer, 2019-02-26:
So on 3.6 I guess they don't do the board voltage check as a prearm check.  I run 3.6 with no problems.  I am running the power module off of 12S but I don't run the ESC from the power module.  The ESC is connected direct to the batteries on a parallel circuit to the power module and servo voltage regulator (stand alone Castle creations 20 Amp voltage regulator).  I configured the power monitor but maybe I configured it incorrectly.

ChristopherOlson, 2019-02-26:
The power monitor configuration should have nothing to do with the Vcc and servo rail voltage. That is internal in the board, based on whatever the voltage is on the board power pins and servo rail. I would check the actual voltage being fed to the board power pins and/or try a different BEC. I see your servo rail is running a 5.2V BEC, maybe wire that up temporary to the spare power plug that comes with the controller and try powering it from that to determine if it's a problem with the power module.

ChristopherOlson, 2019-02-26:
@bnsgeyer I tried 3.7 with a CUAV V5 and I get the same error. PreArm: Board (0.0v) out of range 4.3-5.8v

bnsgeyer, 2019-02-26:
@WickedShell would the hwdef be affected by my selection for the battery monitor? or does it just sense that the pin has got voltage on it?

WickedShell, 2019-02-26:
> So on 3.6 I guess they don't do the board voltage check as a prearm check.

3.6 actually tried to, the problem is that this was a missing feature for any of the ChibiOS based boards then, so you didn't get to take advantage of it.

The HWDEF is expecting `PC0` to be the board voltage monitor. Which matches the listed FMUv5 pinout: https://docs.google.com/spreadsheets/d/1-n0__BYDedQrc_2NHqBenG1DNepAgnHpSGglke-QQwY/edit#gid=912976165 This also matches the schematic listed here: https://github.com/cuav/hardware/blob/master/V5_Autopilot/V5_Ver2/CORE/Pixhack_V5_CORE_V2.12.pdf As long as 5V in is alive then you should be getting power to the pin. (Maybe a configuration/generation problem)

WickedShell, 2019-02-26:
> @WickedShell would the hwdef be affected by my selection for the battery monitor? or does it just sense that the pin has got voltage on it?

HWDEF can change the ADC pin mapping, but for the board voltage it's whatever pin is tagged as `VDD_5V_SENS` in the hwdef, and nothing else should be able to mess that up (unless it's also tagged as GPIO or something and you start mixing modes)

bnsgeyer, 2019-02-26:
@WickedShell  and @tridge   two users get the same issue.  I think this is a valid problem.  How do we proceed?  I wouldn't know where to even start.  Let me know how we can help or if you need someone to test a fix

WickedShell, 2019-02-26:
@tridge will probably be more help from here then I am. I don't have the hardware so I'm unable to replicate the test/result. The docs I can find indicate that this should be a valid feature for those boards, but clearly there is a problem somewhere.

tridge, 2019-02-26:
can you please test on USB power?
With my board on USB I see 4.99V using master, with CUAVv5 firmware

tridge, 2019-02-26:
ahh, I just checked my pre-release dev special CAUVv5 board, and that gives 0.0V. I seem to remember a hardware bug was fixed before final release.

ChristopherOlson, 2019-02-26:
With a production release V5 I show 5.22V and no errors. With my developer edition V5 I get the error either on USB or Power Module.

bnsgeyer, 2019-02-26:
Ok so it is pre production hardware issue that was resolved. Sorry for the run around.

closed 2019-02-26 by bnsgeyer