ArduPilot/ardupilot #10567: ArduPlane + latest JSBSim: Wrong airspeed units conversion

simfinite, 2019-02-19:
### Bug report

**Issue details**

I noticed when running ArduPlane SITL with latest JSBSim that the airspeed value was too low compared to ground speed, even without any wind activated.

It looks like the unit of the airspeed value ('vcas' in the the state/output vector) is knots ([see here](https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/src/input_output/FGOutputFG.cpp#L173)), but it is interpreted as feet/s [in ArduPlane sitl](https://github.com/ArduPilot/ardupilot/blob/cb95bf7814dca40d593a4b86c5a689e6be3bb43f/libraries/SITL/SIM_JSBSim.cpp#L431). Probably, the unit changed at some point after @tridge's fork of JSBSim which became apparent now with the recent update to latest JSBSim. 

The fix is simple, change [libraries/SITL/SIM_JSBSim.cpp:431](https://github.com/ArduPilot/ardupilot/blob/cb95bf7814dca40d593a4b86c5a689e6be3bb43f/libraries/SITL/SIM_JSBSim.cpp#L431) to:
`airspeed = fdm.vcas * KNOTS_TO_METERS_PER_SECOND;`

**Version**
latest

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n.a.

**Hardware type**
n.a.

**Logs**
n.a.

simfinite, 2019-02-23:
PR: https://github.com/ArduPilot/ardupilot/pull/10608

OXINARF, 2019-02-25:
Fixed by #10608, thanks @simfinite!

closed 2019-02-25 by OXINARF