ArduPilot/ardupilot #10586: EKF2: investigate if range finder data is being fused when it shouldn't

rmackay9, 2019-02-21:
[This event](https://discuss.ardupilot.org/t/crash-with-3-6-6-copter/38764) seems to indicate that EKF2 attempted to consume range finder data when the range was far beyond the EK2_RNG_USE_HGT.

As a side note, it is very likely that this vehicle should not have had the EK2_ALT_SOURCE = 1 (Range Finder).  We may want to add a pre-arm check to stop users from setting this parameter to "1" if GPS_TYPE != 0.

Biswajitghosh98, 2019-04-16:
If someone isn't working on it already, can I take up this issue?

rmackay9, 2019-04-16:
@Biswajitghosh98, you're very welcome to take a look at it.  Nobody is actively investigating it at the moment (they may in the future but not right now)

Biswajitghosh98, 2019-04-21:
_fusionModeGPS is a private variable, and I'm finding it difficult to access the same for a pre-arm check. Can you help me how to access the same, or, how to access the GPS_TYPE value?

rmackay9, 2020-11-20:
Here is a similar sounding issue: https://github.com/ArduPilot/ardupilot/issues/13379

peterbarker, 2020-11-25:
Closed by https://github.com/ArduPilot/ardupilot/pull/14375

closed 2020-11-25 by peterbarker