ArduPilot/ardupilot #10590: Tools: Autotest badge is being cached by GitHub.

Williangalvani, 2019-02-21:
The badge being displayed is being cached by GitHub and does not reflect the latest run.
It is mentioned in https://github.com/github/markup/issues/224#issuecomment-33454537 that Apache needs to set the `Cache-Control: no-cache` and `ETag` http headers for the caching to stop.

peterbarker, 2019-02-21:
@Williangalvani I independently noticed that this-morning :-)

I modified the apache config a few hours ago to add the appropriate headers.

Past that it's a matter of waiting for their cache to expire.  A quick google didn't tell me how long that's likely to be.

OXINARF, 2019-02-22:
I've purged GitHub's cache and it's now showing correctly. If someone sees a regression on this, please re-open it.

closed 2019-02-22 by OXINARF