ArduPilot/ardupilot #10598: Rangefinder not being used as altitude source

jwen18, 2019-02-22:
#### Issue details
Using Arducopter 3.6.6 for the Intel Aero, the rangefinder is not being used as the altitude source. We can see the rangefinder is working, but the FC is not using those values for the altitude. EKF2_ALT_SOURCE = rangefinder, EKF2_RNG_USE_HGT = 70%. With the same parameters, the rangefinder (LeddarOne 1D downward-facing lidar) was working fine as the altitude source in Arducopter 3.5.5

#### Platform
[  ] All
[  ] AntennaTracker
[X  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2019-02-22:
In general it's a bad idea to use the range finder as the altitude source.  The vehicle is being flown indoors without a GPS and with a flat floor below it (i.e. no obstacles)?

If there was a log file for 3.5.5 and 3.6.6 that would be a great help.  It would allow us to more quickly determine if it's a regression (which it may very well be).

rmackay9, 2019-02-25:
@jwen18, any chance you can provide the dataflash log?  If not I may close this because support issues should be raised in the forums and then only after they've confirmed to be bugs to we raise them here.

jwen18, 2019-02-27:
Sorry for the late reply. I saw that someone else posted the same issue in the forums, so I will just follow that.

closed 2019-02-27 by jwen18

tareknitro, 2020-03-06:
Hello , did you resolve the problem ? with the old version (3.6-dev) i used the range finder as primary source,indoor at low altitude  it worked fin for me , but now with the new version (3.6.9 ..11)  i use TFMINI  ,it does not work, the sonar range value is correct , but the EKF2 altitude is diferent , is the baro Value , not precise ,not stable ,negative value somtimes .

rmackay9, 2020-03-07:
@tareknitro,

It's probably best to raise this in the forums and include an onboard log.  Ideally it would be best to upgrade to Copter-4.0.3 because this is the version that I'm more actively supporting ([Copter-4.0 support forum](https://discuss.ardupilot.org/c/arducopter/copter-40/155)).  I tested EKF's use of the range finder a couple of months ago and it seemed to be working but we've also got [this report re the TFminiPlus I2C.](https://github.com/ArduPilot/ardupilot/issues/13411).  A log will help us determine if it's related.