ArduPilot/ardupilot #10600: Add RTH height option "at least" (ALT_HOLD_RTL)

TheZippyMan, 2019-02-22:
### Feature request

**Is your feature request related to a problem? Please describe.**
As ChibiOS is getting more popular, many FPV long range guys, including me, start to fly longrange FPV on Ardupilot, because other autopilot firmwares are kind of unstable. Many of those people do not carry a laptop around and care about waypoints or mission planning.
When flying longrange you are manually controlling the plane in (mostly) FBWA mode.
In the case of a failsafe/RTH event the plane does decent to a specified value or stays at the current height if the value "-1" is choosen for ALT_HOLD_RTL. 
This solution is, in my opinion not optimal.
Example: You have set "-1" for ALT_HOLD_RTL and are flying 10m above the ground. In case of a failsafe event the plane will just try to go to the home position and maybe crash.
If a value is set for ALT_HOLD_RTL (example 70m) and you are flying 200m high over some very high trees the plane will decent in a failsafe/RTL case and crash into the trees.

I hope I could explain the issue.

**Describe the solution you'd like**
Add an option "at least".
Plane is at an altitude obove the set height --> will stay there and fly home
Plane is at an altitude below the set height --> will climb and fly home

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2019-02-23:
I think this is what it already does.  Copter does this anyway.. the RTL_ALT is the minimum altitude that it wil fly home at.

TheZippyMan, 2019-02-23:
Okay! I am only flying a plane, didn't know this already exists in copter. Thanks for your response!
Would be great to see this feature in some time.

ninja-zx11, 2019-04-10:
I could be wrong but i think if you are flying at 100m ,4 kms away and your RTL is set to 50m,plane will only start descend after getting close to the home.Right after FS,it will fly at the same altitude it was flying.

WickedShell, 2019-04-10:
@ninja-zx11 This is not true. The vehicle will build a slope from where ever you enter RTL to the home position/altitude. It will then try and fly this line up or down directly. It won't wait until you are close to home to start doing the descent.

ninja-zx11, 2019-04-10:
Oh Okay.I am sorry and i was assuming wrong.Thanks for the clarification.

han8jun, 2020-03-03:
Thoes fpver who like cross trees,brideges can hardly estimate constant height to saftly return.  ALT_HOLD_RTL=-1  is best option, but also need lower limit.
hope this feature implemented at someday.

david9981, 2020-03-04:
If above ALT_HOLD_RTL, maintains current altitude. If below ALT_HOLD_RTL, climb.  This is a very basic but important function. If Copter, and iNav, already has this feature, don't understand why Plane cannot have this.  Please.

peterbarker, 2020-03-04:
Anybody particularly attached to `ALT_HOLD_RTL`?  The Copter equivalent is `RTL_ALT` IIUC.

https://ardupilot.org/copter/docs/parameters.html#rtl-alt-rtl-altitude

Consistency where we can get it is good.

rmackay9, 2020-03-04:
I suspect there will be blockers to changing the parameter name but personally I would love to see it changed to a name starting with "RTL_".  In general we try to put the subsystem or flight mode at the front of the parameter name.

xuxiandi, 2020-03-06:
I think this is s very usefual feature.

xuxiandi, 2020-03-06:
> I suspect there will be blockers to changing the parameter name but personally I would love to see it changed to a name starting with "RTL_". In general we try to put the subsystem or flight mode at the front of the parameter name.
I think this is s very usefual feature.

han8jun, 2020-03-26:
there is a forked code like below. anyone who happen to use omnbusF4 can download the firmware at https://github.com/han8jun/arduplane_atleast_rtl

/*
  return RTL altitude as AMSL altitude
 */
int32_t Plane::get_RTL_altitude()
{
    int32_t rtl_alt=g.RTL_atleast_altitude_cm + home.alt;
    if (g.RTL_altitude_cm < 0) {
        return current_loc.alt;
        if (current_loc.alt>=rtl_alt){
            return current_loc.alt;
        }else{
            return rtl_alt;
        }
    }
    return g.RTL_altitude_cm + home.alt;
}