ArduPilot/ardupilot #10607: Plane: SITL master as of #b290148ed02 cannot Auto Takeoff if terrain_enable is 0

Raviu8, 2019-02-23:
### Bug report
**Issue details**
_Compiled SITL plane as of master branch Today most recent commit is #b290148ed02.  
If I set TERRAIN_ENABLE to 0 the plane cannot takeoff in Auto mode, it accelerates to around 20m/s speed, channel2 output - of elevator - is over 1800 , but it keeps moving on ground forever._

**Version**
_Plane, master branch sitl._

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x ] Plane
[  ] Rover
[  ] Submarine

Raviu8, 2019-02-23:
Mission altitude frame is relative and not terrain.

tridge, 2019-02-23:
I just tried it and it works fine. If you post a log from the logs/ directory perhaps we can work out why it isn't working for you

Raviu8, 2019-02-23:
I've tried too many times and can confirm the behavior.
Here is the log. In this test I've tried to change the takeoff pitch angel three time and restart the mission with no success.
[00000001.zip](https://github.com/ArduPilot/ardupilot/files/2896940/00000001.zip)

Raviu8, 2019-02-23:
Just enable terrain and re-run the same mission, it works fine.

magicrub, 2019-02-23:
@Raviu8 please reset the params with flag -w to ensure you don't have anything funny configured that is messing it up

Raviu8, 2019-02-24:
Sure I've done this many times, even `rm -f eeprom.bin` manually to be 100% sure.
You couldn't reproduce the issue?
I could reproduce it every single time, no exclusions.

peterbarker, 2019-02-24:
I'm also unable to replicate.

I'm starting SITL like this: `./Tools/autotest/sim_vehicle.py -v ArduPlane --map --console`

I loaded your mission.

I armed and changed to auto - it took off OK.

I set the parameter, killed SITL and restarted it.

I armed and changed to auto - it took off.

Can you lay your steps-to-reproduce out, please?

Raviu8, 2019-02-24:
1. clean the ardupilot working dir on master branch
`git clean -f -x -d ; git submodule foreach --recursive "git clean -f -x -d"`
`./waf distclean` 
`rm -fv eeprom.bin` if any
2. build sitl and start it as follows
`./waf configure && ./waf plane && build/sitl/bin/arduplane -M plane -O 42.109,24.955,160,180`

3. Connect to it using MissionPlanner on tcp port, - I use a a windows VM to run MissionPlanner.

4. Set ARMING_CHECK to 14 - to skip INS to be able to arm - and TERRAIN_ENABLE to 0
And write the mission waypoints. 1st point is takeoff - the actual mission used is attached.
[test_takeoff.waypoints.txt](https://github.com/ArduPilot/ardupilot/files/2897693/test_takeoff.waypoints.txt)

5. Kill SITL on terminal with Ctrl+C

6. Run it again `build/sitl/bin/arduplane -M plane -O 42.109,24.955,160,180`
And reconnect on MissionPlanner

7. Arm, change mode to Auto and watch, it would accelerate to around 20m/s but stick to ground forever.

8. Now while it's trying to takeoff, just go to parameters and set TERRAIN_ENABLE to 1, it would instantly takeoff.

_I'm running this version of MissionPlanner ; Just extract it as any compressed archive and run MissionPlanner.exe_

> http://firmware.ardupilot.org/Tools/MissionPlanner/MissionPlanner.appx

peterbarker, 2019-02-26:
Outcomes from devcall:

```
UTC2314 - https://github.com/ArduPilot/ardupilot/issues/10607
Could be default parameter issues
He should try loading default SITL parameters
Reversed elevator?
Could be underground due to terrain data problems?
https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/default_params/plane.parm
```

Raviu8, 2019-02-26:
I'll do another test with the mentioned params.

> Could be underground due to terrain data problems?

Logically terrain data problems should occur when terrain is enabled not disabled, shouldn't it?

> Reversed elevator?

Impossible! Otherwise it would never takeoff and achieve any altitudes correctly. Once terrain is enabled everything is working fine instantly.

WickedShell, 2019-02-26:
Terrain data gets used for some parts of the SITL simulation (ground compass interference at a minimum).

The main point is that default parameter values won't fly correctly, you need to load a parameter file even with SITL before errors are really relevant. (I've watched no param planes think they are going sideways). You should not need to turn off any part of the arming checks. Having to turn these off indicates a configuration problem. (There may be a real problem that needs to be fixed, but until the starting configuration needs to be valid first)

Raviu8, 2019-02-26:
Hi, I've done so many tests using using the default params file with different combination of params to get minimum number of params required to takeoff successfully when terrain_enable = 0.

Results: - of course all when terrain_enable = 0 -
1. Loading full list of default params works fine.
2. INS_ params have no effect on the problem. So I strippem then and added arming_check = 14 to be able to arm.
3. I could takeoff with a complex minimal set of params if any if them is not set, the plane doesn't takeoff.

Here is the final set of params:

> 
> TRIM_ARSPD_CM   2200
> LIM_PITCH_MAX   2500
> ARSPD_USE       1
> THR_MAX		100
> SERVO3_MAX      2000
> 

Also, should set to be able to test:

> TERRAIN_ENABLE  0
ARMING_CHECK    14

IamPete1, 2021-08-15:
I cant reproduce on master.

closed 2021-08-15 by IamPete1