ArduPilot/ardupilot #10612: ArduPlane SITL: JSBSim realtime vs. batch mode

simfinite, 2019-02-24:
### Bug report

**Issue details**

In the process of [investigating time synchronization issues](https://discuss.ardupilot.org/t/arduplane-jsbsim-synchronization-lock-step-mechanism-issues/38781) between ArduPlane SITL and JSBSim, I noticed that JSBSim is currently being run in realtime mode (with the '--realtime' commandline flag). The problem is that in realtime mode,  the ‘iterate 1’ command that is being used to let JSBSim run one frame and suspend afterwards [is not supported](https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/src/JSBSim.cpp#L507). It can be seen from the [JSBSim sources](https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/src/JSBSim.cpp) (lines 515-545) that in realtime mode, more than one frame can be executed each time JSBSim runs.

Although, it did not solve the [synchronization problems discussed here](https://discuss.ardupilot.org/t/arduplane-jsbsim-synchronization-lock-step-mechanism-issues/38781), I believe we should be running JSBSim in batch mode and not in realtime mode. I implemented the [changes required to run JSBSim in batch mode here](https://github.com/simfinite/ardupilot/commits/jsbsim_batchmode) and I have done some preliminary testing. The changes include:

- removed the '--realtime' flag
- new sample rate constant that is used in various places
- added an explicit sleep time of 10us to the '--nice' flag (defaults to 0.01s if none is given, which slows down everything a lot) 

It works well on my setup, but further testing should probably be done. 

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
n.a.

**Hardware type**
n.a.

**Logs**
n.a.

rajat2004, 2019-04-24:
Hi,
I was just going through the [SIM_JSBSim.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_JSBSim.cpp) file when I noticed that the simulation rate had been set as 1000Hz, and not by using a variable or a rate constant
https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_JSBSim.cpp#L215

I'm not sure why this was done, since the default Ardupilot SITL simulation rate is 1200Hz I believe
https://github.com/ArduPilot/ardupilot/blob/master/libraries/SITL/SIM_Aircraft.cpp#L54

While trying to figure this out, I found your post on the forums and this issue. I think that your approach of running JSBSim in batch mode instead of real-time mode is correct since `iterate 1`  is used for implementing lock-step scheduling I believe.

One thing I'm a bit confused about is the [comment](https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/src/JSBSim.cpp#L507) in the JSBSim source file about `iterate` not being supported in real-time mode but only in batch-mode & play-nice mode, aren't there only 2 modes in JSBSim, Realtime & Batch modes with `nice` being an option in both of them? 
This is most probably just a mistake while writing the comments but just wanted to make sure.

I'll try out some of the changes you have made and see what's the difference. Some dev should look into this, if we're going in the right direction or if this is the intended behaviour

simfinite, 2019-04-24:
> aren't there only 2 modes in JSBSim, Realtime & Batch modes with nice being an option in both of them?

I think what the comment is referring to as 'play-nice mode' is the combination of batch mode with the additional `--nice` flag which will make JSBSim sleep for a specified (or default) time after each step. Without this flag it would [just run flat-out fast](https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/src/JSBSim.cpp#L510). In realtime mode the nice flag doesn't really make any sense, since the application should sleep just as long as required to keep realtime, but they may [have implemented additional sleeping in realtime mode](https://github.com/JSBSim-Team/jsbsim/blob/06f4ce37619d73c619c6296d1f602b10088bd95d/src/JSBSim.cpp#L539) as an option to slow things down in this mode as well.

Now, that I think about it, we may not even need the `--nice` flag at all with batch mode and `iterate 1` as we're stopping after each single step anyway. I may give this a try.

rajat2004, 2019-04-24:
> I think what the comment is referring to as 'play-nice mode' is the combination of batch mode with the additional `--nice` flag which will make JSBSim sleep for a specified (or default) time after [...]

Not sure where this kind of combination will be required but seems nice to have that option

> In realtime mode the nice flag doesn't really make any sense, since the application should sleep just as long as required to keep realtime, but they may [have implemented additional sleeping in [...]

Looks like additional options have been kept for all kind of scenarios.
The `sleep_nseconds` parameter for the sleeping time when `playing nice` is [set here](https://github.com/JSBSim-Team/jsbsim/blob/6190fcc32051fd96c601b329ed320a67cacdc725/src/JSBSim.cpp#L495)
So in `realtime` mode, the simulator would sleep for the `frame_duration` which is decided by `simulation_rate` parameter.
In `Batch mode`, it would sleep for the time specified by `nice` flag

> Now, that I think about it, we may not even need the `--nice` flag at all with batch mode and `iterate 1` as we're stopping after each single step anyway. I may give this a try.

That is what I was thinking since we're already specifying the `simulation_rate`, then the `nice` option might not be required at all in Batch_mode.

rajat2004, 2019-04-24:
I tried out the changes which you had made, my branch is here if interested: 
https://github.com/rajat2004/ardupilot/tree/jsbsim-sitl-test

Changed from Realtime to batch mode and also removed the `nice` flag but that slowed everything down which doesn't make sense since removing the `nice` option should speed it up, I'll try to do a more in-depth analysis later. 
Therefore added the `nice` flag with the timestep as 10us as you had mentioned. I felt that the simulation was a bit off-track with this, but it might be that I'm sleepy right now, will try to record video and upload it if that would help

One thing, what's the proper way to shutdown JSBSim, doing Ctrl + C on the terminal keeps the JSBSim process running which I have to manually kill or restart.

rajat2004, 2019-04-24:
https://drive.google.com/open?id=1AZjDj0u9g3XxXQWDV55YmBcYevuAvhBs
Video with the Batch Mode and nice sleeptime set to 10us

https://drive.google.com/open?id=1T6cjtFFsEh1q9zwbApv18V1xvtibU4ID
Video with realtime mode

simfinite, 2019-04-25:
I am having a hard time trying to interpret what exactly I see in the videos. However, there seem to be some differences which is not good. From what I know, the only reasonable explanation for these differences is the [synchronisation issue which I have written about here](https://discuss.ardupilot.org/t/arduplane-jsbsim-synchronization-lock-step-mechanism-issues/38781). 

In a few words: The current approach is to wait for UDP frames sent from JSBSim and incrementing the simulation time in ArduPlane SITL with each frame. Unfortunately, UDP transport is unreliable and packets may be dropped or received multiple times, which leads to a time lag between ArduPlane and JSBSim simulation time, which may lead to all sorts of problems.

The time lag / time drift can be made observable with some [minor modifications in ArduPlane and JSBSim described here](https://discuss.ardupilot.org/t/arduplane-jsbsim-synchronization-lock-step-mechanism-issues/38781/2). Basically, we need JSBSim to transmit the actual simulation time along with each frame and compare it to the ArduPlane time once received.

I have observed this time lag in both batch and realtime mode on my setup. However, it may be that switching to batch mode makes things worse. It could also depend on the individual setup and system load which would be even worse.

@rajat2004 If you could confirm that you also have this time lag, then we could be reasonably sure it's not an issue of my local setup only.

About your question on shutting down JSBSim:
> One thing, what's the proper way to shutdown JSBSim, doing Ctrl + C on the terminal keeps the JSBSim process running which I have to manually kill or restart.

I also have had problems with dangling JSBSim processes ever since we switched to latest JSBSim. Usually I just close the sim_vehicle.py with Ctrl+C. Not sure why it works sometimes and sometimes not. May be worth opening a new issue.

rajat2004, 2019-04-25:
@simfinite I made the changes which you had done for getting the simulation time but I can't see any difference in the output, i.e. I'm not seeing any simulation time data in the CSV file

File attached here for review. This was done with Batch mode and nice sleeptime of 10us
[jsbsim_csv_output.log](https://github.com/ArduPilot/ardupilot/files/3117029/jsbsim_csv_output.log)

My JSBSim branch is here, please check if there're any errors in how I modified it - https://github.com/rajat2004/jsbsim/tree/ardupilot-test
I have pushed the CSV logdirective changes also to my Ardupilot branch mentioned above so please check that also.
 
One thing I'm able to get my head around is why the `nice` parameter is causing this kind of behaviour in Batch mode. It's taking very long to arm and everything's super slow. I tried out 1us and 100us as well but that gave even worse simulation resuts, felt like the plane was going in random directions.

>  However, it may be that switching to batch mode makes things worse. It could also depend on the individual setup and system load which would be even worse.

This shouldn't be happening. From my understanding, lock-step scheduling removes the affect of individual systems by specifying the time-step so that even on faster systems, everyting works at the same speed. But then, nothing's working as expected right now in this.

>  I also have had problems with dangling JSBSim processes ever since we switched to latest JSBSim

Interesting, so it used to exit properly in the older version?

simfinite, 2019-04-25:
> This shouldn't be happening. From my understanding, lock-step scheduling removes the effect of individual systems by specifying the time-step so that even on faster systems, everything works at the same speed.

Correct and it's even more than a speed issue: The lock-step mechanism *should* let the ArduPlane SITL and JSBSim process run in a deterministic forth-and-back scheme, so that simulation results are 100% reproducible. The undeterministic time lag, I have observed, compromises the deterministic behavior of the simulation. 

About the JSBSim CSV logs, I just used these to generate [the plots in my post here](https://discuss.ardupilot.org/t/arduplane-jsbsim-synchronization-lock-step-mechanism-issues/38781). If you just want to observe the time lag, it's sufficient to output the ArduPlane-internal time and the transmitted JSBSim simulation time at the end of [JSBSim::recv_fdm in SIM_JSBSim.cpp](https://github.com/ArduPilot/ardupilot/blob/961085fbd19f3bb0e124b8924a9aed1ad405dd15/libraries/SITL/SIM_JSBSim.cpp#L408), e.g.:
`printf("\ntime lag: %u us, ArduPlane time: %u us , JSBSim time: %u us", (time_now_us-fdm.cur_time), time_now_us, fdm.cur_time);`

Of course that requires the little modification of the UDP protocol in JSBSim, that sets the member `cur_time` of the UDP packet to the simulation time.

> why the nice parameter is causing this kind of behaviour in Batch mode. It's taking very long to arm and everything's super slow. 

This behavior, I have only experienced, when not setting a sleeping time in batch mode explicitely, i.e. providing only a blank `--nice`flag. JSBSim then defaults to a sleeping time of 0.01 seconds, which slows everything down tremendously.

rajat2004, 2019-04-25:
> If you just want to observe the time lag, it's sufficient to output the ArduPlane-internal time and the transmitted JSBSim simulation time at the end of JSBSim::recv_fdm in SIM_JSBSim.cpp, e.g.:
printf("\ntime lag: %u us, ArduPlane time: %u us , JSBSim time: %u us", (time_now_us-fdm.cur_time), time_now_us, fdm.cur_time);

Just tried this out, I'm also getting the time-lag, but it's a bit different from yours
I'm seeing a more rapid time-lag increase when running in real-time (current master)

```
time lag: 3142000 us, ArduPlane time: 11767000 us , JSBSim time: 8625000 us
time lag: 3142000 us, ArduPlane time: 11768000 us , JSBSim time: 8626000 us
time lag: 3142000 us, ArduPlane time: 11769000 us , JSBSim time: 8627000 us
time lag: 3143000 us, ArduPlane time: 11770000 us , JSBSim time: 8627000 us
time lag: 3143000 us, ArduPlane time: 11771000 us , JSBSim time: 8628000 us
time lag: 3143000 us, ArduPlane time: 11772000 us , JSBSim time: 8629000 us
time lag: 3144000 us, ArduPlane time: 11773000 us , JSBSim time: 8629000 us
time lag: 3144000 us, ArduPlane time: 11774000 us , JSBSim time: 8630000 us
time lag: 3144000 us, ArduPlane time: 11775000 us , JSBSim time: 8631000 us
time lag: 3145000 us, ArduPlane time: 11776000 us , JSBSim time: 8631000 us
time lag: 3145000 us, ArduPlane time: 11777000 us , JSBSim time: 8632000 us
time lag: 3146000 us, ArduPlane time: 11778000 us , JSBSim time: 8632000 us
time lag: 3146000 us, ArduPlane time: 11779000 us , JSBSim time: 8633000 us
time lag: 3146000 us, ArduPlane time: 11780000 us , JSBSim time: 8634000 us

```

The following is from running in Batch mode with nice as 10us

```
time lag: 13050001 us, ArduPlane time: 88521000 us , JSBSim time: 75470999 us
time lag: 13050001 us, ArduPlane time: 88522000 us , JSBSim time: 75471999 us
time lag: 13051001 us, ArduPlane time: 88523000 us , JSBSim time: 75471999 us
time lag: 13051001 us, ArduPlane time: 88524000 us , JSBSim time: 75472999 us
time lag: 13051001 us, ArduPlane time: 88525000 us , JSBSim time: 75473999 us
time lag: 13051001 us, ArduPlane time: 88526000 us , JSBSim time: 75474999 us
time lag: 13051001 us, ArduPlane time: 88527000 us , JSBSim time: 75475999 us
time lag: 13051001 us, ArduPlane time: 88528000 us , JSBSim time: 75476999 us
time lag: 13051001 us, ArduPlane time: 88529000 us , JSBSim time: 75477999 us
time lag: 13051001 us, ArduPlane time: 88530000 us , JSBSim time: 75478999 us
time lag: 13051001 us, ArduPlane time: 88531000 us , JSBSim time: 75479999 us
time lag: 13051001 us, ArduPlane time: 88532000 us , JSBSim time: 75480999 us
time lag: 13051001 us, ArduPlane time: 88533000 us , JSBSim time: 75481999 us
time lag: 13051001 us, ArduPlane time: 88534000 us , JSBSim time: 75482999 us
time lag: 13051001 us, ArduPlane time: 88535000 us , JSBSim time: 75483999 us
time lag: 13051001 us, ArduPlane time: 88536000 us , JSBSim time: 75484999 us
time lag: 13051001 us, ArduPlane time: 88537000 us , JSBSim time: 75485999 us
time lag: 13052001 us, ArduPlane time: 88538000 us , JSBSim time: 75485999 us
```

> >   why the nice parameter is causing this kind of behaviour in Batch mode. It's taking very long to arm and everything's super slow.

> This behavior, I have only experienced, when not setting a sleeping time in batch mode explicitely, i.e. providing only a blank --niceflag. JSBSim then defaults to a sleeping time of 0.01 seconds, which slows everything down tremendously.

That's the thing, in the current master, we're not setting the time for the nice flag, so it's set to the `frame_duration` 
`frame_duration` is equal to `dT` which is the inverse of `simulation_rate`, so when we're running this at 1000 Hz in real-time, this sleep time is more than 10us or even 100us but when running it with explicit `nice` sleeptime, it's slowing down which is logically wrong. With such a sleep time, it should be faster, and even faster when not playing nice but that's not happening

> The `sleep_nseconds` parameter for the sleeping time when `playing nice` is [set here](https://github.com/JSBSim-Team/jsbsim/blob/6190fcc32051fd96c601b329ed320a67cacdc725/src/JSBSim.cpp#L495)
> So in [...]

rajat2004, 2019-04-26:
> so when we're running this at 1000 Hz in real-time, this sleep time is more than 10us or even 100us but when running it with explicit nice sleeptime in Batch mode, it's slowing down which is logically wrong. With such a sleep time, it should be faster, and even faster when not playing nice but that's not happening

I was wrong in this, running in Batch mode without specifying the sleeptime using nice, i.e. just using plain `--nice` slows the whole thing down like you said @simfinite 
I ran it with a sleeptime of 10ns, that didn't slow down at all, the time-lag is still there, but it's longer now, taking more time to increase

```
time lag: 2868000 us, ArduPlane time: 23843000 us , JSBSim time: 20975000 us
time lag: 2868000 us, ArduPlane time: 23844000 us , JSBSim time: 20976000 us
time lag: 2868000 us, ArduPlane time: 23845000 us , JSBSim time: 20977000 us
time lag: 2868000 us, ArduPlane time: 23846000 us , JSBSim time: 20978000 us
time lag: 2868000 us, ArduPlane time: 23847000 us , JSBSim time: 20979000 us
time lag: 2868000 us, ArduPlane time: 23848000 us , JSBSim time: 20980000 us
time lag: 2868000 us, ArduPlane time: 23849000 us , JSBSim time: 20981000 us
time lag: 2868000 us, ArduPlane time: 23850000 us , JSBSim time: 20982000 us
time lag: 2868000 us, ArduPlane time: 23851000 us , JSBSim time: 20983000 us
time lag: 2868000 us, ArduPlane time: 23852000 us , JSBSim time: 20984000 us
time lag: 2868000 us, ArduPlane time: 23853000 us , JSBSim time: 20985000 us
time lag: 2868000 us, ArduPlane time: 23854000 us , JSBSim time: 20986000 us
time lag: 2868000 us, ArduPlane time: 23855000 us , JSBSim time: 20987000 us
time lag: 2868000 us, ArduPlane time: 23856000 us , JSBSim time: 20988000 us
time lag: 2868000 us, ArduPlane time: 23857000 us , JSBSim time: 20989000 us
time lag: 2868000 us, ArduPlane time: 23858000 us , JSBSim time: 20990000 us
time lag: 2868000 us, ArduPlane time: 23859000 us , JSBSim time: 20991000 us
time lag: 2868000 us, ArduPlane time: 23860000 us , JSBSim time: 20992000 us
time lag: 2868000 us, ArduPlane time: 23861000 us , JSBSim time: 20993000 us
time lag: 2868000 us, ArduPlane time: 23862000 us , JSBSim time: 20994000 us
time lag: 2868000 us, ArduPlane time: 23863000 us , JSBSim time: 20995000 us
time lag: 2868000 us, ArduPlane time: 23864000 us , JSBSim time: 20996000 us
time lag: 2869000 us, ArduPlane time: 23865000 us , JSBSim time: 20996000 us
time lag: 2869000 us, ArduPlane time: 23866000 us , JSBSim time: 20997000 us
time lag: 2870000 us, ArduPlane time: 23867000 us , JSBSim time: 20997000 us
time lag: 2870000 us, ArduPlane time: 23868000 us , JSBSim time: 20998000 us
time lag: 2870000 us, ArduPlane time: 23869000 us , JSBSim time: 20999000 us
time lag: 2870000 us, ArduPlane time: 23870000 us , JSBSim time: 21000000 us
time lag: 2870000 us, ArduPlane time: 23871000 us , JSBSim time: 21001000 us
time lag: 2870000 us, ArduPlane time: 23872000 us , JSBSim time: 21002000 us
time lag: 2870000 us, ArduPlane time: 23873000 us , JSBSim time: 21003000 us
time lag: 2870000 us, ArduPlane time: 23874000 us , JSBSim time: 21004000 us
time lag: 2870000 us, ArduPlane time: 23875000 us , JSBSim time: 21005000 us
time lag: 2870000 us, ArduPlane time: 23876000 us , JSBSim time: 21006000 us
time lag: 2870000 us, ArduPlane time: 23877000 us , JSBSim time: 21007000 us
time lag: 2870000 us, ArduPlane time: 23878000 us , JSBSim time: 21008000 us
time lag: 2870000 us, ArduPlane time: 23879000 us , JSBSim time: 21009000 us
time lag: 2870000 us, ArduPlane time: 23880000 us , JSBSim time: 21010000 us
time lag: 2870000 us, ArduPlane time: 23881000 us , JSBSim time: 21011000 us
time lag: 2870000 us, ArduPlane time: 23882000 us , JSBSim time: 21012000 us
time lag: 2871000 us, ArduPlane time: 23883000 us , JSBSim time: 21012000 us

```

Like you said, this is probably due to the synchronisation issue. But the time-lag behaviour is better than real-time atleast when running in Batch mode. One problem is why do we still need to use the `nice` option in Batch mode, not using that slows down the simulation which is opposite of it's intended use.

simfinite, 2019-04-26:
Thank you @rajat2004 for reproducing the time lag issue. 

I can confirm that running batch mode without the nice flag, slows down the simulation a lot. No idea why, looking at the implementation in JSBSim.cpp it shouldn't. For now, just setting the flag and specifying a small enough sleeping time may just suffice, although I really would like to know why...

rajat2004, 2019-04-26:
Yeah, so seems like we need the `nice` flag after all in Batch mode as well. What would be a good enough time-step? I tried out 10ns, the time-lag log is above.
One more thing, is the time-lag worse on your system when running in Batch mode? I noticed an improvement in my system when switching to Batch mode atleast for the time-lag, but the performance in the simulation was worse as seen in the video

simfinite, 2019-04-26:
Yes, I have seen the time lag increase faster in batch mode than in realtime mode in a virtual machine setup. But I never could tell much of a difference in the simulated aircraft's behavior, i.e. the time lag increase seemed to be slow enough not to completely mess up the Kalman filters etc. 

What happens when you resolve the time lag issue [like it was done here](https://github.com/simfinite/ardupilot/blob/8485904142b46f2c0255852b1d0a93482fcb664a/libraries/SITL/SIM_JSBSim.cpp#L414) (changes in JSBSim::recv_fdm should be all)?  Do you still get "bad performance"?

rajat2004, 2019-04-26:
> Yes, I have seen the time lag increase faster in batch mode than in realtime mode in a virtual machine setup. But I never could tell much of a difference in the simulated aircraft's behavior, i.e. the time lag increase seemed to be slow enough not to completely mess up the Kalman filters etc.

This is different from mine, on my setup, in Batch mode, the time-lag increased slowly, not always but most of the time it took more time than real-time. I have pasted the logs above

> What happens when you resolve the time lag issue [like it was done here](https://github.com/simfinite/ardupilot/blob/8485904142b46f2c0255852b1d0a93482fcb664a/libraries/SITL/SIM_JSBSim.cpp#L414) (changes in JSBSim::recv_fdm should be all)? Do you still get "bad performance"?

Yes, better than without the patch but still bad

rajat2004, 2019-04-26:
I'm running this on a native Linux system btw

rajat2004, 2019-04-27:
@simfinite I just noticed right now that I didn't apply your patch entirely, as in I forgot to make this change to `time_now_us` :-  `time_now_us = fdm.cur_time;`

But I tried out a different method, I didn't set `time_now_us` to `fdm.curr_time`, I kept it as earlier i.e. 
`time_now_us += 1e6/SIM_RATE_HZ;`
Correspondingly, in your patch, I changed the condition `while (fdm.cur_time == time_now_us);` to 
`while (fdm.cur_time <= time_now_us);`

I'm not getting any lag now at all and simulation is working perfectly. The following is the time-lag log
```
time lag: 0 us, ArduPlane time: 160099000 us , JSBSim time: 160099000 us
time lag: 0 us, ArduPlane time: 160100000 us , JSBSim time: 160100000 us
time lag: 0 us, ArduPlane time: 160101000 us , JSBSim time: 160101000 us
time lag: 0 us, ArduPlane time: 160102000 us , JSBSim time: 160102000 us
time lag: 0 us, ArduPlane time: 160103000 us , JSBSim time: 160103000 us
time lag: 0 us, ArduPlane time: 160104000 us , JSBSim time: 160104000 us
time lag: 0 us, ArduPlane time: 160105000 us , JSBSim time: 160105000 us
time lag: 0 us, ArduPlane time: 160106000 us , JSBSim time: 160106000 us
time lag: 0 us, ArduPlane time: 160107000 us , JSBSim time: 160107000 us
time lag: 0 us, ArduPlane time: 160108000 us , JSBSim time: 160108000 us
time lag: 0 us, ArduPlane time: 160109000 us , JSBSim time: 160109000 us
time lag: 0 us, ArduPlane time: 160110000 us , JSBSim time: 160110000 us
time lag: 0 us, ArduPlane time: 160111000 us , JSBSim time: 160111000 us
time lag: 0 us, ArduPlane time: 160112000 us , JSBSim time: 160112000 us
time lag: 0 us, ArduPlane time: 160113000 us , JSBSim time: 160113000 us

```

A video of the simulation with my work-around, Batch Mode with nice sleeptime as 1us
https://drive.google.com/open?id=10ABrzSFT1msepzU2UMNihuohT_lkZMD4

My Ardupilot branch with the fix: https://github.com/rajat2004/ardupilot/tree/jsbsim-test2

Can you have a look at this?

rajat2004, 2019-04-27:
@tridge If possible, could you have a look also and check which is the correct method to go about this?

rajat2004, 2019-04-27:
@tridge @simfinite We have defined the `SIM_RATE_HZ` parameter which has been used everywhere but shouldn't we be using the `rate_hz` variable defined in `SIM_Aircraft`
https://github.com/ArduPilot/ardupilot/blob/e6c7fd2acc18a0976b1800b15aa1670d948c6d8c/libraries/SITL/SIM_Aircraft.h#L169

rajat2004, 2019-04-28:
If this approach of going with Batch mode instead of Realtime mode is correct then should we open a PR for using the `curr_time` parameter in JSBSim which will be required.
Or is there any other method which doesn't require modifying the JSBSim source code?

rajat2004, 2019-04-28:
@simfinite I think that a PR can be opened for this now, depending on the devs reviews, the corresponding PR for JSBSim can be opened
Should I create the PR for this with my fix?

simfinite, 2019-04-28:
I would rather wait for @tridge or any other dev to decide whether or not we should fix the synchronization issue *before* having JSBSim run in batch mode or if we can already go ahead and take care of the synchronization issue afterwards. At least, I would like to know if all tests pass in batch mode and simulation looks ok on various setups/machines.

If we decide to try and get the JSBSim fix into the JSBSim repo, then we could wait for that to happen and have both the switch to batch mode and the synchronization fix handled in a single pull request that will definitely be an improvement on all existing setups.

About the simulation rate define/variable:
> shouldn't we be using the rate_hz variable defined in SIM_Aircraft

Probably, but I don't know. Right now the rate is hard-coded. I came up with the SIM_RATE_HZ just because I needed it in several places, but if there is already such a variable we should probably use it.

About the synchronization fix:
From what I've seen your implementation yields the same result as mine. Both ways work. Keep in mind that both fixes require the change in JSBSim's UDP protocol.

simfinite, 2019-04-28:
... let's see what the JSBSim people think of it.

rajat2004, 2019-04-28:
> I would rather wait for @tridge or any other dev to decide whether or not we should fix the synchronization issue _before_ having JSBSim run in batch mode or if we can already go ahead and take care [...]

Sounds good, it would be nice if this could be tested on other setups as well. But I don't think there are any tests as such for checking the working of other simulators
> 
> If we decide to try and get the JSBSim fix into the JSBSim repo, then we could wait for that to happen and have both the switch to batch mode and the synchronization fix handled in a single pull request that will definitely be an improvement on all existing setups.

I think it would be better to have everything in a single PR, but as you said depends on the dev's review

> About the simulation rate define/variable:
>
>
> Probably, but I don't know. Right now the rate is hard-coded. I came up with the SIM_RATE_HZ just because I needed it in several places, but if there is [...]
 
The thing is, I noticed afterwards that this variable isn't being used in other places also, such as in the Morse backend. There might be some reason as to why this hasn't been used anywhere else.

Nice, you opened the JSBSim issue for using the `curr_time` field, let's see if they are fine with this

rajat2004, 2019-04-28:
Seems they are okay with it

rajat2004, 2019-04-29:
Should we first try to get a review for this approach before finalizing and merging the JSBSim PR? 
I think that we'll need to open a PR for this here in order to get this reviewed

simfinite, 2019-04-29:
You're probably right. If you have some time on your hand, feel free to go ahead and make a pull request.

rajat2004, 2019-04-29:
@simfinite Any ideas about why using Batch mode with `nice` is causing such slowdown? Maybe an issue for this can be opened in JSBSim, see if they are aware of this

simfinite, 2019-04-30:
No, I currently have no theory why this is. Before opening an issue in the JSBSim repo, we should be sure that the problem is not on ArduPlane side of things.

rajat2004, 2019-04-30:
I'll do a more thorough testing and codebase review after a few days(have exams right now), see what's happening behind the scenes, in Ardupilot as well as JSBSim

rajat2004, 2019-05-10:
The PR has been merged, this issue can be closed now I think
@simfinite A new issue for the problems while exiting JSBSim?

simfinite, 2019-05-10:
Thanks @rajat2004 for all the help and the PR. I'll open another one for the JSBSim shutdown issue.

closed 2019-05-10 by simfinite