ArduPilot/ardupilot #10617: Copter: Auto's Land command initialises target position incorrectly

rmackay9, 2019-02-25:
Copter-3.6.6 has an issue in which a Land mission command perhaps incorrectly initialises the target position which leads to a sudden change in the vehicle's target velocity and acceleration.  In one log this led to a target 45degree lean angle on a vehicle that could not safely lean this far over leading to a crash.

It is possible that the last waypoint command is not being treated as a "slow waypoint".  With "slow waypoints" the vehicle should not consider it's current waypoint complete until it is within the WP_RADIUS of the target position.

peterbarker, 2019-02-25:
On Sun, 24 Feb 2019, Randy Mackay wrote:

> Copter-3.6.6 has an issue in which a Land mission command perhaps incorrectly initialises the target position which leads to a sudden change in the
> vehicle's target velocity and acceleration. In one [...]

precision landing or not?

rmackay9, 2019-02-25:
@peterbarker, it probably affects all situations where we switch from straight line waypoints to Land commands.

rmackay9, 2019-03-05:
This is fixed in master and will go out with Copter-3.6.7

closed 2019-03-05 by rmackay9