ArduPilot/ardupilot #10622: Plane: Add voltage scaling to compensate for voltage drop (as in copter FW)

mikeo0, 2019-02-25:
### Feature request

Add battery voltage scaling to throttle to compensate for dropping battery voltage throughout the flight. 

Especially when using Li-Ion batteries, and no airspeed sensor, the airspeed will drop significantly during an Automous mission. For simple set-ups and for smaller airframes, where AS sensor is not an option, a throttle compensation will allow for the plane to cruise at it's most efficient AS in the entire fligth and deplete the batteries fully,

A battery voltage compensation as already implemented in Copter FW would be able to do this.

I did experiment with using ESCs in govenor mode, but they seem to interact poorly with Arduplane, and you dont get a consistent throttle setting.

This is for Plane FW

estebanavv, 2019-03-16:
Maybe an option would be to implement a throttle by wire that uses RPM feedback as the reference, so 50% throttle would be (max RPMs)/2 , this might require an additional setup step to determine the RPM range for the throttle range input.

giacomo892, 2020-08-20:
It has been implemented. This one can be closed. https://github.com/ArduPilot/ardupilot/pull/14187

closed 2020-08-20 by magicrub

mikeo0, 2020-08-21:
Great, thanks!

Sendt fra Mail til Windows 10

Fra: giacomo892
Sendt: 20. august 2020 15:29
Til: ArduPilot/ardupilot
Cc: mikeo0; Author
Emne: Re: [ArduPilot/ardupilot] Plane: Add voltage scaling to compensatefor voltage drop (as in copter FW) (#10622)

It has been implemented. This one can be closed. #14187