ArduPilot/ardupilot #10623: EKF control limits sometimes fail to limit vehicle movements

peterbarker, 2019-02-25:
### Bug report

**Issue details**

Copter' OpticalFlowLimits test sometimes fails in CI

Vehicle goes faster than the flow limits should allow - speed seems unbounded by the limits.  The effects of this on a real aircraft could be rather nasty.

This test does not fail on my local machine.  I haven't seen it fail on the autotest server either.

Logs are uploaded to our autotest server now, so we should be able to rescue one of them from CI.

The problem may be reproducible with

`./Tools/autotest/autotest.py build.ArduCopter fly.ArduCopter.OpticalFlowLimits`

Diagnosis might be possible by creating a WIP PR which CI can then run.  Debug can then be added as appropriate.

**Version**

master

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QuadCopter

**Hardware type**
SITL

rmackay9, 2019-02-25:
related issue: https://github.com/ArduPilot/ardupilot/issues/10624

peterbarker, 2019-02-25:
#10624 failures are unrelated

peterbarker, 2026-11-15:
Ancient.  This one hasn't flaked for years.

closed 2026-11-15 by peterbarker