ArduPilot/ardupilot #10641: extend version info to include time/date of build

robertschulman1, 2019-02-26:
### Feature request

Developers may build a number of build images from a given version of AP source. In order to distinguish between these builds, I have added identification info to the version string which, during AP operation, will uniquely identify the build of AP in use. The unique information is the time/date info of when AP was built (this assumes no more than one build of AP will be built for a developer in a second).

Rather than using time/date as the unique identifier, I could have used a monotonic counter. This counter would be problematic in a distributed build environment. Time and date is a bit crude, but it is practical and universal in its approach while providing the required uniquification.

A patch file with the suggested changes is attached.
[d.txt](https://github.com/ArduPilot/ardupilot/files/2906536/d.txt)

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

The changes to Copter to could easily be scaled to other platforms.

WickedShell, 2019-02-26:
Why do you need the date? The normal version of identification is the git hash, which has the advantage of being able to go directly from a logfile/build to the exact version of source in a single step.

peterbarker, 2019-02-26:
This PR was seeking to use a field in AUTOPILOT_VERSION to get the information to the GCS: https://github.com/ArduPilot/ardupilot/pull/7287 

I'm pretty sure that PR is going to get bounces, but I'm going to mark it DevCallTopic to either get agreement that we want the feature or that we won't - in which case I can kill the PR.

robertschulman1, 2019-02-27:
My experience in software dev’t tells me that build info (how-things-were-built as opposed to what-was-built) is vital info. My belief is that date/time info is a reasonable proxy for how-things-were-built for those who wish to track such things.

 

bob

IamPete1, 2021-08-29:
Closed by https://github.com/ArduPilot/ardupilot/pull/7287, build with `--build-dates`

closed 2021-08-29 by IamPete1