ArduPilot/ardupilot #10646: Azure/Cygwin autotest problems

OXINARF, 2019-02-27:
Azure is running autotest on Cygwin once per day, but there are, at least, two issues:

- as said on the initial PR, mavutil TCP 3 attempts to reconnect isn't always enough:
  >@tridge @peterbarker Autotest periodically fails in Cygwin/Azure due to MAVProxy only trying to reconnect 3 times after a SITL reboot (https://github.com/ArduPilot/pymavlink/blob/master/mavutil.py#L1147). I tried finding a way to increase this (or even better, make it keep retrying like it does with serial) but it doesn't look easy, any hints?

  Read https://github.com/ArduPilot/ardupilot/pull/10027#pullrequestreview-185409252 and https://github.com/ArduPilot/ardupilot/pull/10027#issuecomment-448256615 for more discussion on it.

- there is a problem with QuadPlane test, it fails due to the code below:
https://github.com/ArduPilot/ardupilot/blob/47d8435cd6b31a4069598e58e87a8c286a26a678/Tools/autotest/common.py#L1767-L1769 Error in Azure is [TypeError: not enough arguments for format string](https://dev.azure.com/ardupilot-org/ardupilot/_build/results?buildId=190&view=logs&jobId=996f6018-1d8e-58be-c8e2-5d12b565d06c&taskId=27d82589-f3c9-5506-5667-ba43b16c45a1&lineStart=898&lineEnd=903&colStart=1&colEnd=50) (line 898)

  Indeed the syntax is invalid, but that's an easy fix I'll PR soon.

  More interesting is the reason why it actually fails the test: we are setting home to an altitude of 643100mm and it returns an altitude of 643090mm. The why is clear: when converting to meters, in float it can't represent that number exactly. However, we could do a better job on our side:
https://github.com/ArduPilot/ardupilot/blob/47d8435cd6b31a4069598e58e87a8c286a26a678/libraries/GCS_MAVLink/GCS_Common.cpp#L3686 Is it worth it trying to round it before converting to an int32?

WickedShell, 2019-02-27:
> Is it worth it trying to round it before converting to an int32?

It will probably make some GCS's rework their code, but not many. I actually do the same kind of check the test is trying to do, but due to the above rounding issues I actually replicate the cast -> cast back to compare it. I can update my check I think but this is the sort of thing that I've deployed to work around this. (Was also really frustrating to get right)

peterbarker, 2019-02-27:
On Wed, 27 Feb 2019, Francisco Ferreira wrote:

> there is a problem with QuadPlane test, it fails due to the code below:
>     https://github.com/ArduPilot/ardupilot/blob/47d8435cd6b31a4069598e58e87a8c286a26a678/Tools/autotest/common.py#L1767-L1769 [...]

I found and patched that one yesterday... I've created a PR for it.

OXINARF, 2019-02-28:
@WickedShell 

> It will probably make some GCS's rework their code, but not many.

Humm, why would a GCS need to rework their code? If we round the value before the cast it should be closer to the value a GCS sent.

> I actually do the same kind of check the test is trying to do, but due to the above rounding issues I actually replicate the cast -> cast back to compare it.

Right, but that's not a very good solution because a simple change on ArduPilot's side and your cast may give a different result. I think what GCS/autotest needs to do is allow a 1 cm margin.

@peterbarker Great! Test stills fails due to rounding issue obviously. It might be transient but [Rover failed two tests](https://dev.azure.com/ardupilot-org/ardupilot/_build/results?buildId=210&view=logs&jobId=8bf8aff3-550c-5802-a498-17e0675aa26d&taskId=2adcfd81-f2d6-5482-766c-ab831803738c&lineStart=1&lineEnd=1&colStart=1&colEnd=1).

peterbarker, 2019-02-28:
On Thu, 28 Feb 2019, Francisco Ferreira wrote:

> @peterbarker Great! Test stills fails due to rounding issue obviously. It might be transient but Rover failed two tests.

The not-getting-home is transient, and probably a real bug.

peterbarker, 2019-02-28:
On Thu, 28 Feb 2019, Francisco Ferreira wrote:

> @peterbarker Great! Test stills fails due to rounding issue obviously. It might be transient but Rover failed two tests.

... the plane not coming back after rebooting for rc12 to take effects 
might be a significant issue, however.

OXINARF, 2019-03-01:
> ... the plane not coming back after rebooting for rc12 to take effects might be a significant issue, however.

@peterbarker That's just the identified bug of mavutil only trying to connect 3 times to a TCP port.

IamPete1, 2021-09-12:
Azure is gone, https://github.com/ArduPilot/ardupilot/pull/17251

closed 2021-09-12 by IamPete1