ArduPilot/ardupilot #10650: GPS: investigate GPS week roll-over event

rmackay9, 2019-02-28:
This is an issue to capture the investigation (and any possible code fixes) required to deal with the GPS roll-over event that will happen on April 6th/7th 2019.

There's a bit of info from the UBlox manually below:
[image]

Also there are many discussions on the internet [including this one](https://portal.u-blox.com/s/question/0D52p00008HKDTlCAP/gps-week-number-rollover-after-20-years).

Two things we should probably test:

1. any bad effects of the GPS week rolling over while the vehicle is in the air?  the roll-over will probably happen at midnight GMT time which is 5pm Los Angeles (April 6th?) and 9am Tokyo time (April 7th?),
2. is the vehicle's time set correctly after the roll-over?  ArduPilot sets its "real time" based on the GPS date.  If the calculation is incorrect we can imagine that the dates recorded in log files could be incorrect

rmackay9, 2019-02-28:
I'm pretty confident that from ArduPilot's point of view, it will not see any roll-over in the GPS week unless the vehicle has a very very old GPS (from before drones were invented).

The GPS week rolls over from 1024 to 0 but in our dataflash logs we see that UBlox is reporting the week as 2041 (for this particular GPS).  The UBlox GPSs will internally handle the roll-over so the week number will not suddenly decrease.
[image]

WickedShell, 2019-03-11:
This should be fine.

closed 2019-03-11 by WickedShell

gmorph, 2019-03-31:
From my limited understanding UBlox  GPSs will change from sending 2041 week number to 2042 which isn't magical in any way so ArduPilot should be fine with it. 

The GPS world has already been through this in 1999 so they are aware and know what they should do and have published strategy in place - in theory.

One issue is if UBlox doesn't do that and does start sending 0 - hard for us to test a GPS UBlox for this - and the whole world would have a huge issue as this would be a bug in UBlox's GPS firmware.

What do other GPS manufacturers (not UBlox) that we support do?  

And generic NMEA data stream - how does that one work - ZDA, RMC messsages are dd,mm,yyyy and ddmmyy respectively so the GPS Firmware needs to do right thing - the message itself is fine. 

Should we test when the week rolls over from 2047 to 2048?  We aren't storing the GPS week as an 11bit number so that should be fine too.

Further to this UBlox GPSs and probably others appear to have a 20 year life span so at some point in the not too distant future (years not decades) failures will start happening.  ArduPilot started 2010ish so if people bought GPSs made in 2008 then 2028ish.  Hmmm - few years off yet I guess.

Whilst UBlox have released a statement describing how they handle the rollover I can't find any for the other GPS manufacturers we support.  SwiftNavigation, SiRF (now Qualcomm), 

Septentrion has one:
https://septentrio-my.sharepoint.com/personal/marketing_septentrio_com/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fmarketing%5Fseptentrio%5Fcom%2FDocuments%2FPublic%2FSupport%20Notification%20GPS%20week%20rollover%20March19%2Epdf&parent=%2Fpersonal%2Fmarketing%5Fseptentrio%5Fcom%2FDocuments%2FPublic&cid=ddc88835-ce49-4ec9-950e-df9a94d52f33

MediaTek has one
https://labs.mediatek.com/en/faq/FAQ18906

rmackay9, 2019-03-31:
For anyone else reading this thread, I've created a [blog post here](https://discuss.ardupilot.org/t/gps-week-roll-over-april-6th/40459) which includes a video of some testing of AP in the simulator when the GPS week rolls back to zero and it seems fine.