ArduPilot/ardupilot #10662: Perform prearm checks simultaneously

SergeyBokhantsev, 2019-02-28:
### Feature request

**Is your feature request related to a problem? Please describe.**
There are about 20 pre-arm checks in Copter and Plane at the moment. But they're performed in a sequential way. That means user cannot see all the problems together, instead he must check and solve pre-arm blockers one-by-one. This is a time consuming bustle.
It will be good to see all the blockers simultaneously: Safety switch + Compass inconsistent + Fence + Battery etc.

**Describe the solution you'd like**
1. Always do the prearm check for all systems.
2. Report via onboard OSD (the [Annunciator panel ](https://en.wikipedia.org/wiki/Annunciator_panel)is a good example of how this can looks like)
[image]

3. Report flags to GCS so it can also visualize an annunciator panel.
4. Reporting via GCS message can be slightly improved too by showing the description of the first blocker + total number of blockers pending. And by cycling prearm errors.

**Platform**
- All

WickedShell, 2019-02-28:
> 1. Always do the prearm check for all systems

We do this already. The problem you can see is normally a system will return the first failure it has. So if you have multiple problems with the GPS for example you only see the first one print. But if you have a problem with both the GPS and the safety button for example both will print out.

peterbarker, 2019-02-28:
On Thu, 28 Feb 2019, Sergey Bokhantsev wrote:

> There are about 20 pre-arm checks in Copter and Plane at the moment. But they're performed in a sequential way. That means user cannot see all the
> problems together, instead he must check and solve [...]

This is something I've been moving towards for a long time :-)

I have a branch here which does this: 
https://github.com/peterbarker/ardupilot/tree/arming-check-function

It's 8000 commits behind master, so it won't be rebased, at a guess.

>  1.
>
>     Always do the prearm check for all systems.

commit 015b4d967e034694bf3df3f60e62b994e0f60ddd 
(github/always-do-arming-checks, always-do-arming-checks)
Author: Peter Barker <pbarker@barker.dropbear.id.au>
Date:   Tue Aug 23 13:33:06 2016 +1000

     AP_Arming: prearm checks fill in bitmask of failures

(on a branch, "always-do-arming-checks" :-)

>  2.
>
>     Report via onboard OSD (the Annunciator panel is a good example of how this can looks like)
>     default

This would work like normal statustexts, with normal dwell times etc.

>  3.
>
>     Report flags to GCS so it can also visualize an annunciator panel.

Harder, as adding checks becomes problematic.  We have had discussions on 
the matter....

>  4.
>
>     Reporting via GCS message can be slightly improved too by showing the description of the first blocker + total number of blockers pending. And by
>     cycling prearm errors.

Display issue :-)

SergeyBokhantsev, 2019-03-01:
@peterbarker thx I will look at. 
@WickedShell could you plz clarify how this works? (cannot check on a real HW at the moment)
here the checks are performs sequentially https://github.com/ArduPilot/ardupilot/blob/70d4d2ac8f482e9fa903537a9f8d9d226c9e93fc/libraries/AP_Arming/AP_Arming.cpp#L708

WickedShell, 2019-03-01:
@SergeyBokhantsev that's a bitwise and, not a logical one. That means there is no lazy evaluation here, and all the functions will be called. Inside the specific function something may only do a single check, but all of these checks on the linked line will be called, regardless of if one of the earlier checks failed.

peterbarker, 2024-07-31:
I think this problem has been solves - we do a lot of && stuff in AP_Arming now!

Closing

closed 2024-07-31 by peterbarker