ArduPilot/ardupilot #10678: ADC Pin 15 in Pixhawk 2.1 treated as divided by 2

suraqis, 2019-03-01:
### Bug report
When voltage applied to PIN 15 (ADC port) on Pixhawk 2.1 Cube it is treated as divided by 2 (so measured voltage value is multiplied by 2).

I connected custom temperature sensor to ADC port (PIN 15) and set port as sonar input. I see measured 
 voltage value (sonarvoltage) doubled against measured by a voltmeter.

In practice, it has a simple workaround - to set a range measurement coefficient half of a real one. But I suspect that for any other application it may be confusing. 

**Version**
Arduplane 3.9.5

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Pixhawk Cube

**Logs**
Easily reproducible

omeralierdemir, 2020-06-19:
Hi suraqis, How to read temperature sensor to ADC port ? Which adc port did you use ? Is pin 15 adc pin on the front of pixhawk? where is PIN 15 adc?  I tried to read analog data on ADC pin for temperature sense.
I followed the steps on this site one by one: https://ardupilot.org/plane/docs/common-rangefinder-maxbotix-analog.html. 
I read the sonar voltage indicator values ​​steadily 1.19, but when I throw the sensation of temperature in pixhawk, I keep reading the analog data.  
Did you set a different parameter other than those on this site?

IamPete1, 2021-08-29:
It there is a issue I'm not really sure we can fix it now in anycase, that would break it for everyone who has set the scale to compensate.