ArduPilot/ardupilot #10684: ArduCopter linking error

dashpritam, 2019-03-03:
Hi All !
I am getting a linking error while running ArduCopter SITL on Ubuntu 16.04.

```
[533/533] Linking build/sitl/bin/arducopter
ArduCopter/ArduCopter.cpp.27.o: In function Copter::ten_hz_logging_loop()': ArduCopter.cpp:(.text._ZN6Copter19ten_hz_logging_loopEv+0x12d): undefined reference toAC_PosControl::write_log()’
ArduCopter/Attitude.cpp.27.o: In function Copter::set_throttle_takeoff()': Attitude.cpp:(.text._ZN6Copter20set_throttle_takeoffEv+0x8): undefined reference toAC_PosControl::init_takeoff()’
ArduCopter/GCS_Mavlink.cpp.27.o: In function Copter::send_nav_controller_output(mavlink_channel_t)': GCS_Mavlink.cpp:(.text._ZN6Copter26send_nav_controller_outputE17mavlink_channel_t+0x89): undefined reference toAC_PosControl::get_alt_error() const’
ArduCopter/GCS_Mavlink.cpp.27.o: In function GCS_MAVLINK_Copter::handle_command_long_packet(__mavlink_command_long_t const&)': GCS_Mavlink.cpp:(.text._ZN18GCS_MAVLINK_Copter26handle_command_long_packetERK24__mavlink_command_long_t+0x530): undefined reference toAC_WPNav::set_speed_up(float)’
GCS_Mavlink.cpp:(.text._ZN18GCS_MAVLINK_Copter26handle_command_long_packetERK24__mavlink_command_long_t+0x6c1): undefined reference to AC_WPNav::set_speed_xy(float)' GCS_Mavlink.cpp:(.text._ZN18GCS_MAVLINK_Copter26handle_command_long_packetERK24__mavlink_command_long_t+0x711): undefined reference toAC_WPNav::set_speed_down(float)’
ArduCopter/mode_althold.cpp.27.o: In function `Copter::ModeAltHold::run()’:
```

I checked make.inc and AC_PosControl and AC_WPNAV are included in the Libraries field. Please share your recommendations if anyone have come across similar issue.

Thanks,

**Version**
Linux 16.04

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
Issue in SITL 

**Logs**
_Please provide a link to any relevant logs that show the issue_

IamPete1, 2019-03-03:
try a dist clean and re-build

rmackay9, 2019-03-04:
Maybe try a "git submodule init" and "git submodule update --recursive"

You've probably already seen these but just in case:

- [setting up the build environment on Ubuntu/Linux](http://ardupilot.org/dev/docs/building-setup-linux.html)
- [gitter chat](https://gitter.im/ArduPilot/ardupilot)

I think I'll close this because it seems to be building on my virtual linux box

closed 2019-03-04 by rmackay9

nabeelsherazi, 2021-01-12:
Hey @dashpritam did you ever figure out what caused this issue? I'm getting the same error and this issue is the only result on google. Dist clean and re-build isn't doing anything.

rmackay9, 2021-01-13:
@nabeelsherazi,

Some other things that may help:

- delete the "build" directory.  I think this is normally within the "ardupilot" directory but it might be higher up
- make sure the packages required to build are up-to-date.  lxml caused some problems about a week ago because a dependency was added to the build that required the latest lxml release but some developers (including me) didn't have it installed and there is no check in the build that all the right packages are installed, instead it just fails with a hard-to-understand error message.  I don't think lxml will be the cause of your issue but perhaps another package is out-of-date.