ArduPilot/ardupilot #10688: Disable roll and pitch stabilization

jlopez1337, 2019-03-03:
#### Issue details
I'm trying to disable the roll and pitch stabilization in stabilize mode of the copter.
There seems to be some complicated smoothing calculations for stabilizing them. I tried to disable
the calculations for them in the method thrust_heading_rotation_angles in AC_AttitudeControl.cpp, but it also causes weird behavior for the controller pitch and roll inputs (I don't want to disable it for controller inputs).
Here are the changes I made (taking only the target values for roll and pitch)
```cpp
void AC_AttitudeControl::thrust_heading_rotation_angles(Quaternion& att_to_quat, const Quaternion& att_from_quat, Vector3f& att_diff_angle, float& thrust_vec_dot)
{
    Matrix3f att_to_rot_matrix; // rotation from the target body frame to the inertial frame.
    att_to_quat.rotation_matrix(att_to_rot_matrix);
    Vector3f att_to_thrust_vec = att_to_rot_matrix*Vector3f(0.0f,0.0f,1.0f);

    Matrix3f att_from_rot_matrix; // rotation from the current body frame to the inertial frame.
    att_from_quat.rotation_matrix(att_from_rot_matrix);
    Vector3f att_from_thrust_vec = att_from_rot_matrix*Vector3f(0.0f,0.0f,1.0f);

    // the cross product of the desired and target thrust vector defines the rotation vector
    Vector3f thrust_vec_cross = att_from_thrust_vec % att_to_thrust_vec;
    Vector3f thrust_vec_cross_rp = att_to_thrust_vec;

    // the dot product is used to calculate the angle between the target and desired thrust vectors
    thrust_vec_dot = acosf(constrain_float(att_from_thrust_vec * att_to_thrust_vec,-1.0f,1.0f));
    float thrust_vec_dot_rp = acosf(constrain_float(att_to_thrust_vec*Vector3f(1,1,1), -1.0f, 1.0f));

    // Normalize the thrust rotation vector
    float thrust_vector_length = thrust_vec_cross.length();
    if(is_zero(thrust_vector_length) || is_zero(thrust_vec_dot)){
        thrust_vec_cross = Vector3f(0,0,1);
        thrust_vec_dot = 0.0f;
    }else{
        thrust_vec_cross /= thrust_vector_length;
    }

    if(is_zero(thrust_vector_length) || is_zero(thrust_vec_dot_rp)){
        thrust_vec_cross_rp = Vector3f(0,0,1);
        thrust_vec_dot_rp = 0.0f;
    }else{
        thrust_vec_cross_rp /= thrust_vector_length;
    }

    Quaternion thrust_vec_correction_quat;
    thrust_vec_correction_quat.from_axis_angle(thrust_vec_cross, thrust_vec_dot);

    // Rotate thrust_vec_correction_quat to the att_from frame
    thrust_vec_correction_quat = att_from_quat.inverse()*thrust_vec_correction_quat*att_from_quat;

    Quaternion thrust_vec_correction_quat_rp;
    thrust_vec_correction_quat_rp.from_axis_angle(thrust_vec_cross_rp, thrust_vec_dot_rp);
    thrust_vec_correction_quat_rp = att_from_quat.inverse()*thrust_vec_correction_quat_rp*att_from_quat;

    // calculate the remaining rotation required after thrust vector is rotated transformed to the att_from frame
    Quaternion yaw_vec_correction_quat = thrust_vec_correction_quat.inverse()*att_from_quat.inverse()*att_to_quat;

    // calculate the angle error in x and y.
    Vector3f rotation;
    thrust_vec_correction_quat_rp.to_axis_angle(rotation);
    att_diff_angle.x = rotation.x;
    att_diff_angle.y = rotation.y;

    // calculate the angle error in z (x and y should be zero here).
    yaw_vec_correction_quat.to_axis_angle(rotation);
    att_diff_angle.z = rotation.z;

    // Todo: Limit roll an pitch error based on output saturation and maximum error.

    // Limit Yaw Error based on maximum acceleration - Update to include output saturation and maximum error.
    // Currently the limit is based on the maximum acceleration using the linear part of the SQRT controller.
    // This should be updated to be based on an angle limit, saturation, or unlimited based on user defined parameters.
    if(!is_zero(_p_angle_yaw.kP()) && fabsf(att_diff_angle.z) > AC_ATTITUDE_ACCEL_Y_CONTROLLER_MAX_RADSS/_p_angle_yaw.kP()){
        att_diff_angle.z = constrain_float(wrap_PI(att_diff_angle.z), -AC_ATTITUDE_ACCEL_Y_CONTROLLER_MAX_RADSS/_p_angle_yaw.kP(), AC_ATTITUDE_ACCEL_Y_CONTROLLER_MAX_RADSS/_p_angle_yaw.kP());
        yaw_vec_correction_quat.from_axis_angle(Vector3f(0.0f,0.0f,att_diff_angle.z));
        att_to_quat = att_from_quat*thrust_vec_correction_quat*yaw_vec_correction_quat;
    }
}
```

Also, I changed the inputs the code in rate_controller_run

```cpp
void AC_AttitudeControl_Multi::rate_controller_run()
{
    // move throttle vs attitude mixing towards desired (called from here because this is conveniently called on every iteration)
    update_throttle_rpy_mix();

    Vector3f gyro_latest = _ahrs.get_gyro_latest();
    _motors.set_roll(rate_target_to_motor_roll(0, _rate_target_ang_vel.x));
    _motors.set_pitch(rate_target_to_motor_pitch(0, _rate_target_ang_vel.y));
    _motors.set_yaw(rate_target_to_motor_yaw(gyro_latest.z, _rate_target_ang_vel.z));

    control_monitor_update();
}
```

#### Version
Copter V3.5.7
MissionPlanner V1.3.62

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

#### Hardware type
Pixhawk1 / px4-v2 (stm32f427VI).

IamPete1, 2019-03-03:
you could set the gains to zero, or use acro mode, seems like quite a odd thing to want to do

kd0aij, 2019-03-04:
What are you trying to accomplish?

closed 2021-07-26 by IamPete1