ArduPilot/ardupilot #10689: Add rm3100 geomagnetometer i2c

roque-canales, 2019-03-03:
Critical point for antenna tracker is its compass.

There is new technology named geomagnetometer wich is far more reliable and precis thans embeded magnetometer on pixhawk boards.

With this new compass our tracker will track morer precisely uavs with less error allowing the use of larger db high gain antennas

auturgy, 2019-03-03:
ArduPilot has a driver for RM3100, although currently it is only available in “master”. I expect it to be included in the next round of stable releases.
https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_Compass

closed 2019-03-03 by auturgy

omegaretro, 2020-05-18:
I compiled a 4.0.4 master for the pixhawk1-1m, but the rm3100 is not being listed in the compass_typemask which may have been canceled during the compilation appears the rm3100 being compiled