ArduPilot/ardupilot #10717: SITL on Windows (Cygwin) broken

hamishwillee, 2019-03-06:
**Issue details**

SITL on Windows native installation as per http://ardupilot.org/dev/docs/sitl-native-on-windows.html fails with `RiTW: Window access not found, logging to /tmp/ArduCopter.log`.

The error is as reported in https://github.com/ArduPilot/ardupilot/issues/7897 so is at least a year old. 
```
...
'build' finished successfully (15.357s)
SIM_VEHICLE: Using defaults from (/home/hamis/ardupilot/Tools/autotest/default_params/copter.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/hamis/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/hamis/ardupilot/build/sitl/bin/arducopter" "-S" "-I0" "--home" "-35.363261,149.165230,584,353" "--model" "+" "--speedup" "1" "--defaults" "/home/hamis/ardupilot/Tools/autotest/default_params/copter.parm"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "/usr/bin/cygstart" "-w" "mavproxy.exe" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551"
RiTW: Starting ArduCopter : /home/hamis/ardupilot/build/sitl/bin/arducopter -S -I0 --home -35.363261,149.165230,584,353 --model + --speedup 1 --defaults /home/hamis/ardupilot/Tools/autotest/default_params/copter.parm
which: no osascript in (/usr/local/bin:/usr/bin:/cygdrive/c/Program Files (x86)/Common Files/Oracle/Java/javapath:/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/Program Files (x86)/Intel/iCLS Client:/cygdrive/c/Program Files/Intel/iCLS Client:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/cygdrive/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/cygdrive/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/IPT:/cygdrive/c/Program Files/Intel/Intel(R) Management Engine Components/IPT:/cygdrive/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/cygdrive/c/WINDOWS/system32:/cygdrive/c/WINDOWS:/cygdrive/c/WINDOWS/System32/Wbem:/cygdrive/c/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/d/Program Files (x86)/Calibre2:/cygdrive/d/Program Files/Git/cmd:/cygdrive/d/Program Files (x86)/WinMerge:/cygdrive/c/WINDOWS/System32/OpenSSH:/cygdrive/d/apps/HashiCorp/Vagrant/bin:/cygdrive/d/Program Files (x86)/MAVProxy:/cygdrive/c/Program Files/Microsoft SQL Server/130/Tools/Binn:/cygdrive/c/Program Files (x86)/Windows Kits/8.1/Windows Performance Toolkit:/cygdrive/d/Program Files (x86)/MAVProxy:/cygdrive/c/Program Files/nodejs:/cygdrive/c/Users/hamis/AppData/Local/Programs/Python/Python37-32/Scripts:/cygdrive/c/Users/hamis/AppData/Local/Programs/Python/Python37-32:/cygdrive/d/Ruby25-x64/bin:/cygdrive/c/Users/hamis/AppData/Local/Microsoft/WindowsApps:/usr/bin:/cygdrive/c/Program Files/heroku/bin:/cygdrive/c/Users/hamis/AppData/Local/Pandoc:/cygdrive/c/Users/hamis/AppData/Roaming/npm:/home/hamis/ardupilot/Tools/autotest:/cygdrive/c/pixhawk_toolchain/toolchain/bin)
RiTW: Window access not found, logging to /tmp/ArduCopter.log
SIM_VEHICLE: MAVProxy exited
SIM_VEHICLE: Killing tasks
/home/hamis/ardupilot/Tools/autotest/run_in_terminal_window.sh: line 54:  4874 Killed                  "$cmd" $* &> "$filename" < /dev/null
```

Note that I added osascript, xterm, konsole, gnome-terminal as documented indicated in https://github.com/ArduPilot/ardupilot/issues/7897#issuecomment-417809195

The arducopter.log is unhelpful:
```
Set parameter SIM_SPEEDUP to 1.000000
Creating model + at speed 1.0
Home: -35.363261 149.165230 alt=584.000000m hdg=353.000000
Starting sketch 'ArduCopter'
Starting SITL input
Using Irlock at port : 9005
bind port 5760 for 0
Serial port 0 on TCP port 5760
Waiting for connection ....
```

**Version**
Master 6 March 2019

**Platform**

Copter SITL

**Airframe type**

Default

**Hardware type**

SITL

hamishwillee, 2019-03-06:
@auturgy As an aside, this means the only way I can get SITL working on windows is in a VM or dronekit-sitl old build. Neither cygwin or vagrant build working.

I am trying to verify that the tuning page displays in QGC 3.5 for latest Copter SITL  because it doesn't on Ubuntu VM. It does if ArduPilot is running on Chibi OS, or using an old version of Copter from DK-SITL. If you happen to have this working on Windows in a recent build would be great to have confirmation that this does work!

magicrub, 2019-03-06:
I'll try on today's master, but I know it worked a few weeks ago..

OXINARF, 2019-03-06:
SITL works on Cygwin. We are building it on two CIs and running autotest daily on Azure - it run today one hour after this was opened.

Not having a window for SITL is a sim_vehicle.py message, has nothing to do with SITL running or not. And it doesn't stop sim_vehicle.py from starting SITL as proved by the log.

You should check your MAVProxy installation, that's what isn't showing any output.

Anyway, please don't open issues unless it's a confirmed bug, our list is already enormous. If you are having issues, please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot), that's what they are there for.

auturgy, 2019-03-06:
Just because this works on the CI server doesn’t make it universally good. I started working through this with Hamish last night. The Cygwin install script definitely didn’t result in a working setup on my Windows box, so we need to fix the script, the instructions, or both.

magicrub, 2019-03-06:
>..Cygwin install script..

@auturgy  What script? I have [this one](https://github.com/magicrub/ardupilot/commit/e5ce21ec0bc20b73f03c1183734be294c0b7cf02) in a branch but it needs a lot of work before it could be merged.

magicrub, 2019-03-06:
[image]

magicrub, 2019-03-06:
I'll leave this open for now, but I'm 99.9% sure this should be moved to a wiki issue and not ardupilot issue.

auturgy, 2019-03-06:
This script: https://raw.githubusercontent.com/ArduPilot/ardupilot/master/Tools/autotest/win_sitl/InstallDevEnvironmentAndAPMSource.ps1

Regards,

James

hamishwillee, 2019-03-06:
@auturgy Both ideally, but by preference the script. 

@magicrub 

> I'll leave this open for now, but I'm 99.9% sure this should be moved to a wiki issue and not ardupilot issue.

I posted here deliberately. Having a robust end-user-developer environment setup script is so important it belongs in the platform and should be tested as part of CI. The wiki should just reference a robust setup script/tool.

OXINARF, 2019-03-06:
@auturgy 

> Just because this works on the CI server doesn’t make it universally good.

No, it doesn't, but the issues list isn't the place for each individual problem each user has. "Just because this doesn't work for you, it doesn't make it broken"... If our own team can't understand this is unmanageable, how will users?

> I started working through this with Hamish last night. The Cygwin install script definitely didn’t result in a working setup on my Windows box, so we need to fix the script, the instructions, or both.

Instructions are in the wiki that, as you know, has its own repo for issues. The OP doesn't mention anywhere that script was used nor that any issue was found with it. In fact, nothing was mentioned about it being a clean setup, the title even was `SITL on Windows (Cygwin) broken`, nothing to do with setup.

What the OP says:
> SITL on Windows native installation as per http://ardupilot.org/dev/docs/sitl-native-on-windows.html fails with RiTW: Window access not found, logging to /tmp/ArduCopter.log.

As I said in my post this is no failure/error; in fact, it is absolutely normal. As I also said there is no indication that SITL is broken, the shown log is fine.
The only indication present is the MAVProxy exiting and I'm sure you also know this isn't the MAVProxy repo.

I'm sorry but this is exactly the type of issue that stays open for ages without any action on it - and the reason is simple: the only way to try to do something about it is someone reinstalling their build system in a clean system and even then it's not guaranteed it is reproducible.
Instead, it would have been super easy to drop by Gitter and ask for help figuring out what is wrong - then create an issue in the appropriate place(s) with what the problem is/what needs to change.

hamishwillee, 2019-03-07:
@OXINARF There is a little picture process issue here and a big picture ArduPilot issue. The little picture is that I could have  posted this elsewhere or worked through it on gitter. Yes, I didn't mention the script - because I didn't know it existed.

The big picture is that this is you think of this as a "Hamish" problem, and I *know* that actually this is a serious ArduPilot problem: the dev team has not taken ownership of creating a robust developer toolchain for Windows.

Had I posted this in the wiki area the issue would have still sat around forever, the only difference is that the people who might work to sort out the big picture would have been able to ignore it.

rmackay9, 2019-03-07:
I'll try and have a look at this although I might need to ask Tridge or Peter for help..

This issue is probably in the wrong place but Hamish is senior enough that we can put that aside until we solve the problem. 

I'll remove the cygwin instructions from [the offending wiki page](http://ardupilot.org/dev/docs/sitl-native-on-windows.html) and instead point users back to [this cygwin setup page](http://ardupilot.org/dev/docs/building-setup-windows-cygwin.html).

hamishwillee, 2019-03-07:
Thanks @rmackay9 - I'll try that other page today! Probably overstating this, but even if it works, would be great if this was scripted and the script was run in CI.

rmackay9, 2019-03-07:
I've just pushed some [changes to the wiki page](http://ardupilot.org/dev/docs/sitl-native-on-windows.html).  It was suffering from a lot of duplicate and out-of-date info I think.  It works for me but let's see.

hamishwillee, 2019-03-07:
@rmackay9 Thanks for that. I have got this working by _almost_ following your _precise_ instructions (almost). The problem is indeed MAVProxy as indicated by @OXINARF. 

@stephendade FYI

I had to do two things:
1. Path was not being set. Had to be manually added.
1. First time around I took the latest, but actually low 1.5.x is the only one that works
   - 1.5.2 [here](http://firmware.ardupilot.org/Tools/MAVProxy/) is the last version that works robustly on my machine. 
   - 1.5.7 - 1.6.x versions install and start, but fail shortly after with various errors:
      ```
      Exception in thread main_loop:
      Traceback (most recent call last):
      File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\threading", line 810, in __bootstrap_inner
      File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\threading", line 763, in run
      File "<string>", line 801, in main_loop
      File "<string>", line 528, in process_master
      File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.dialects.v20.ardupilotmega", line 9689, in parse_buffer
      File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.dialects.v20.ardupilotmega", line 9625, in parse_char
      File "C:\Users\Stephen\Documents\MAVProxy\MAVProxy\build\mavproxy\out00-PYZ.pyz\pymavlink.dialects.v20.ardupilotmega", line 9602, in __callbacks
      File "D:\Program Files (x86)\MAVProxy\eggs\mavproxy-1.5.0-py2.7.egg\MAVProxy\modules\mavproxy_link.py", line 398, in master_callback
      self.report_altitude(m.relative_alt*0.001)
      File "D:\Program Files (x86)\MAVProxy\eggs\mavproxy-1.5.0-py2.7.egg\MAVProxy\modules\mavproxy_link.py", line 243, in report_altitude
      math.fabs(self.status.altitude - self.status.last_altitude_announce) >= AttributeError: 'MPStatus' object has no attribute 'last_altitude_announce'
   - 1.7.x and later start for a fraction of a second and then die (no useful logs).

Further thoughts 
1. Docs are much improved by @rmackay9  recent change, in particular the automated selection of dependencies to cygwin
1. I still get missing osascript messages, which look bad, but don't appear to affect operation.
1. Mavproxy map uses tiles from OviHybrid provider (at least for this version) - but those don't give me any map. Not such a great first-user experience on start.
1. Still get "RiTW: Window access not found, logging to /tmp/ArduCopter.log". Again, bit ugly but not stopping operation.

I'd still like a script that runs in CI and tests this end to end :-)

stephendade, 2019-03-07:
@hamishwillee - Interesting! MAVProxy 1.8.4 works fine on my Win10 PC. Are you able to grab some output from it to diagnose your issue? I find executing MAVProxy in a batch file with a ``pause`` command at the end allows me to see any error messages before MAVProxy exits.

For the MAVProxy path, that should be added as part of the install process for the last couple of versions.

In terms of the general issue being discussed, I agree that we need to merge together and have a single point of reference in both the install install scripts and documentation. CI by Appveyor would be nice too - I'm generally a big fan all "test all the things in CI".

I'd suggest:
- Go through the documentation and have just the one page (for each OS) for installing the enviroment. Both the Windows and Linux sections for http://ardupilot.org/dev/docs/building-the-code.html and http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html duplicate the install instructions.
- Ensure there's only a single script(s) for installing the dev environment for each OS in the Ardupilot repo. Cuts down on complexity.
- Tweak the Appveyor CI to use the same install scripts as the user

I'm happy to take a look at it over the weekend

khancyr, 2019-03-07:
@stephendade I completly agree for the wiki rewrite ! I have made a bunch rewrite for linux install, if windows part could be cleaned, it would be very nice !

hamishwillee, 2019-03-07:
Hi @stephendade,

Docs plan sounds excellent - in particular including SITL and other dev env setup for each OS in one page. Possible exception is worth having separate pages for IDEs.

If I run 1.8.4 just in a command prompt I get the following output:
```
C:\Users\hamis>mavproxy.exe
Traceback (most recent call last):
  File "MAVProxy\mavproxy.py", line 12, in <module>
  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 395, in load_module
  File "site-packages\serial\__init__.py", line 27, in <module>
  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 395, in load_module
  File "site-packages\serial\serialwin32.py", line 15, in <module>
  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 395, in load_module
  File "site-packages\serial\win32.py", line 21, in <module>
  File "site-packages\PyInstaller\loader\pyiboot01_bootstrap.py", line 174, in __init__
__main__.PyInstallerImportError: Failed to load dynlib/dll 'D:\\Program Files (x86)\\MAVProxy\\kernel32'. Most probably this dynlib/dll was not found when the application was frozen.
[3368] Failed to execute script mavproxy
```

stephendade, 2019-03-08:
@hamishwillee - could you try the MAVProxy installer here: https://ci.appveyor.com/project/stephendade/mavproxy/builds/22921549/artifacts

hamishwillee, 2019-03-08:
@stephendade Same problem
```
Traceback (most recent call last):
  File "site-packages\PyInstaller\loader\pyiboot01_bootstrap.py", line 172, in __init__
  File "ctypes\__init__.py", line 351, in __init__
OSError: [WinError 126] The specified module could not be found

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "MAVProxy\mavproxy.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "C:\Python35\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
  File "site-packages\serial\__init__.py", line 27, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "C:\Python35\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
  File "site-packages\serial\serialwin32.py", line 15, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "C:\Python35\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
  File "site-packages\serial\win32.py", line 21, in <module>
  File "site-packages\PyInstaller\loader\pyiboot01_bootstrap.py", line 174, in __init__
__main__.PyInstallerImportError: Failed to load dynlib/dll 'D:\\Program Files (x86)\\MAVProxy\\kernel32'. Most probably this dynlib/dll was not found when the application was frozen.
[23688] Failed to execute script mavproxy
```

stephendade, 2019-03-08:
@hamishwillee ... Can you check if you have the " Visual C++ Redistributable for Visual Studio 2015 " installed? If not you can download it from https://www.microsoft.com/en-US/download/details.aspx?id=48145

Also, could you confirm the _exact_ Windows version you're running?

stephendade, 2019-03-09:
So, I've done a few more patches: https://ci.appveyor.com/project/tridge/mavproxy/builds/22945498/artifacts

@hamishwillee - I've tried the above installer on a fresh Win10 VM with no issues.

hamishwillee, 2019-03-12:
Hi @stephendade ,

Thanks for that. I tried the latest build - same issue:
```
Traceback (most recent call last):
  File "MAVProxy\mavproxy.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "C:\Python35\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
  File "site-packages\serial\__init__.py", line 27, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "C:\Python35\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
  File "site-packages\serial\serialwin32.py", line 15, in <module>
  File "<frozen importlib._bootstrap>", line 968, in _find_and_load
  File "<frozen importlib._bootstrap>", line 957, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "C:\Python35\lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 627, in exec_module
  File "site-packages\serial\win32.py", line 21, in <module>
  File "site-packages\PyInstaller\loader\pyiboot01_bootstrap.py", line 174, in __init__
__main__.PyInstallerImportError: Failed to load dynlib/dll 'D:\\Program Files (x86)\\MAVProxy\\kernel32'. Most probably this dynlib/dll was not found when the application was frozen.
[16784] Failed to execute script mavproxy
```

1. Yes I do have the redistributable
   [image]
2. The version is:
   [image]
   And updated to the latest today
   [image]

Is there any debugging I could do locally? Worse case I could bring this laptop to Canberra :-)

stephendade, 2019-03-12:
> Go through the documentation and have just the one page (for each OS) for installing the enviroment. Both the Windows and Linux sections for http://ardupilot.org/dev/docs/building-the-code.html and [...]

Turns out @rmackay9 had already cleaned up most of the issues with duplicated install instructions. I've gone through the Wiki and cleaned up a few things (https://github.com/ArduPilot/ardupilot_wiki/pull/1677). One that goes in, the documentation side of things should be good.

hamishwillee, 2019-03-13:
Hi Stephen,

Thanks for your help.

I tested 1.8.4 on another Windows 10 computer and it worked fine. 

I tried your instructions above on my "problem" computer, and the exe that was built to MAVProxy\MAVProxy\dist\mavproxy works fine :-(. So not sure what we can do to progress then even if you had this computer?

Just FYI, I think your dev env setup instructions in this post for are much simpler than in http://ardupilot.github.io/MAVProxy/html/development/mavdevenvwindows.html#windows-development-environment - might be worth considering updating?

hamishwillee, 2019-03-19:
I'm closing this because the instructions are much improved and work on one of my computers. I'll see if I can bring this computer to Stephen on the weekend, but either way this could reasonably be opened as a MAVProxy rather than Firmware issue.

Thank you all for your help.

closed 2019-03-19 by hamishwillee

PrasadNR, 2019-10-21:
Could anyone help me with https://github.com/ArduPilot/ardupilot/pull/12006#issuecomment-544586009 ?