ArduPilot/ardupilot #10718: Vagrant SITL build and connect issues

hamishwillee, 2019-03-06:
**Issue details**

Vagrant build of ArduCopter from latest master running on Windows 10 . Setup according to http://ardupilot.org/dev/docs/setting-up-sitl-using-vagrant.html

There are two issues. 

1. When building there is an error at the end:

```
   bionic64: config.status: executing depfiles commands
    bionic64: config.status: executing libtool commands
    bionic64: configure: WARNING: unrecognized options: --enable-maintainer-mode, --enable-compile-warnings
    bionic64: Now type `make' to compile JSBSim
    bionic64: + make -j2
    bionic64: Making all in src
    bionic64: make[1]: Entering directory '/home/vagrant/jsbsim/src'
    bionic64: Making all in initialization
    bionic64: make[2]: Entering directory '/home/vagrant/jsbsim/src/initialization'
    bionic64:   CXX      FGTrim.o
    bionic64:   CXX      FGInitialCondition.o
    bionic64:   CXX      FGTrimAxis.o
    bionic64:   CXX      FGSimplexTrim.o
    bionic64:   CXX      FGTrimmer.o
    bionic64: In file included from ../../src/math/FGStateSpace.h:27:0,
    bionic64:                  from FGSimplexTrim.h:23,
    bionic64:                  from FGSimplexTrim.cpp:20:
    bionic64: ../../src/models/propulsion/FGTurbine.h:297:3: error: ‘FGParameter’ does not name a type; did you mean ‘FGThruster’?
    bionic64:    FGParameter *N1SpoolUp;
    bionic64:    ^~~~~~~~~~~
    bionic64:    FGThruster
    bionic64: ../../src/models/propulsion/FGTurbine.h:298:3: error: ‘FGParameter’ does not name a type; did you mean ‘FGThruster’?
    bionic64:    FGParameter *N1SpoolDown;
    bionic64:    ^~~~~~~~~~~
    bionic64:    FGThruster
    bionic64: ../../src/models/propulsion/FGTurbine.h:299:3: error: ‘FGParameter’ does not name a type; did you mean ‘FGThruster’?
    bionic64:    FGParameter *N2SpoolUp;
    bionic64:    ^~~~~~~~~~~
    bionic64:    FGThruster
    bionic64: ../../src/models/propulsion/FGTurbine.h:300:3: error: ‘FGParameter’ does not name a type; did you mean ‘FGThruster’?
    bionic64:    FGParameter *N2SpoolDown;
    bionic64:    ^~~~~~~~~~~
    bionic64:    FGThruster
    bionic64: ../../src/models/propulsion/FGTurbine.h:310:1: error: expected class-name before ‘{’ token
    bionic64:  {
    bionic64:  ^
    bionic64: Makefile:445: recipe for target 'FGSimplexTrim.o' failed
    bionic64: make[2]: *** [FGSimplexTrim.o] Error 1
    bionic64: make[2]: *** Waiting for unfinished jobs....
    bionic64: make[2]: Leaving directory '/home/vagrant/jsbsim/src/initialization'
    bionic64: Makefile:612: recipe for target 'all-recursive' failed
    bionic64: make[1]: Leaving directory '/home/vagrant/jsbsim/src'
    bionic64: make[1]: *** [all-recursive] Error 1
    bionic64: Makefile:387: recipe for target 'all-recursive' failed
    bionic64: make: *** [all-recursive] Error 1
The SSH command responded with a non-zero exit status. Vagrant
assumes that this means the command failed. The output for this command
should be in the log above. Please read the output to determine what
went wrong.
```

2. Can't connect to the environment from QGC. For UDP just sits forever. For TCP reports error below:

[image]

peterbarker, 2019-03-07:
TCP is never going to work; the instance isn't listening on `127.0.0.1` (not even close...)

Please paste a screenshot of the configuration of the UDP port.

hamishwillee, 2019-03-08:
Do you mean this?
[image]

peterbarker, 2019-03-08:
@hamishwillee I was referring to the QGC configuration screen :-)

Please retry the Vagrant machine bringup using master.

peterbarker, 2019-03-08:
... mavproxy on the host machine can receive the `:14550` traffic.

hamishwillee, 2019-03-12:
Fixed by https://github.com/ArduPilot/ardupilot/pull/10741

Thanks @peterbarker !!

closed 2019-03-12 by hamishwillee