ArduPilot/ardupilot #10719: Copter SITL does not display tuning page in QGC

hamishwillee, 2019-03-06:
**Issue details**

QGC 3.5 connected to latest master Copter SITL build does not display the tuning page. Tested on Ubuntu 16 (VM) and Windows (Cygwin build run from DK-SITL). 

The tuning page is displayed for Chibi OS builds, and for older versions of ArduPilot (e.g. Copter 3.3), so probably Copter problem rather than QGC issue.

hamishwillee, 2019-03-12:
FYI @DonLakeFlyer @rmackay9 . Note, just retested. See no tuning page on copter. Have not tested on Sub or other platforms. Because this works on older Copter (in DK-SITL) I am posting as an ArduPilot rather than QGC issue.

[image]

DonLakeFlyer, 2019-03-22:
Have you setup an airframe?

hamishwillee, 2019-03-25:
@DonLakeFlyer Don't need to - that is done by SITL. 
But you can see that it is set here:
[image]

IamPete1, 2021-08-15:
On master we can select the tuning page but there is nothing on it.
[image]

And we get this exciting error:

[image]

If I set `MNT_TYPE` so the mnt params are no longer hidden the error drops down to:

[image]

I'm not really sure what we can do about this from the AP side, so I tend to think we should close....