ArduPilot/ardupilot #10752: Rover: Cannot enter Guided mode with only wheel-encoders

rmackay9, 2019-03-09:
Rover-3.5.0 apparently doesn't allow the user to enter Guided mode if only Wheel Encoders are being used for the position estimate.  The cause is likely to do with the new EKF failsafe code that was added to 3.5.  The EKF failsafe check likely checks the **absolute** position estimate is good when it should also work when only the **relative** position estimate is good.

This was reported by @jmachuca77 and @magicrub.

peterbarker, 2019-03-09:
On Fri, 8 Mar 2019, Randy Mackay wrote:

> Rover-3.5.0 apparently doesn't allow the user to enter Guided mode if only Wheel Encoders are being used for the position estimate. The cause is likely to
> do with the new EKF failsafe code that was [...]

I don't think this problem is a result of any EKF changes.  The MAVLink 
API seems to convert everything into llh.

It would be nice to clarify the callbacks on the modes.

Perhaps "requires_llh" ("requires_global_position"?) and 
"requires_position"?  Unfortunately, this is a touch simplistic as the 
answer is "sometimes".  If the user wants to use 
MAVLINK_MSG_ID_SET_POSITION_TARGET_LOCAL_NED they will not need (e.g.) a 
GPS lock.  If they use MAVLINK_MSG_ID_SET_POSITION_TARGET_GLOBAL_INT they 
will - and you won't know what they want before entering the mode.  So 
that probably means you enter guided if you can sense relative position 
but need to bounce any attempt to move you in the global FOR.

jmachuca77, 2019-03-10:
It's a result of this change commit 80e9a54
Before it did not care about EKF failsafe which not sure if it's correct but it allowed us to go into guided mode.

rmackay9, 2019-03-11:
We should be able to do what we do in Copter.  A relative position source is actually OK because we still force the user to set an EKF origin (and thus home) so the vehicle knows where it is in the world.  There will be some drift but that's OK.

Rajbhanu1, 2019-04-26:
Any update on this issue. as we are struggling hard to use my rover with encoder. Any direction or help is appreciated

rmackay9, 2019-10-17:
I think I've got a fix for this issue in this branch: https://github.com/rmackay9/rmackay9-ardupilot/commits/wenc-fix

rmackay9, 2019-10-22:
This is fixed in master so closing

closed 2019-10-22 by rmackay9