ArduPilot/ardupilot #10771: UKF (Unscented Kalman Filter) instead of EKF (Extended Kalman Filter) ?

MohamedAliRashad, 2019-03-10:
### Feature request

I have been going throw some self driving courses and i found that a lot of people prefer **UKF** over **EKF** when it comes to performance so i was wondering if it was considered upgrading to **UKF** ? 

The idea of **UKF** is basically the same as **EKF** but instead of linearizing the model with only one point and approximating the mean and variance based on it, we take sigma points with weights for each point and get a better Gaussian approximation.

Hope this idea can be considered for future releases.

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

magicrub, 2019-03-11:
From what I remember, UKF has I much higher CPU demand. Now that F7 is around, and even double precision hardware, this might be a viable option to pursue. Do you have experience with it? Our library is set up so that it can be added in parallel with existing systems. See EKF2 and EKF3 and DCM.

olliw42, 2019-03-11:
AFAIK AutoQuad is (was, project seems dead) using UKF since many years, on very similar F4 hardware.

MohamedAliRashad, 2019-03-11:
@magicrub 
Well, i am actually going through the implementation of UKF right now with Radar and Lidar data but this is for cars not copters ... i would like to know first what is the state vector used in Ardupilot and how the EKF code was written here.

bluuu81, 2019-03-11:
I'm voting for add a UKF. In AQ it did a job without any problems.
MCU utilization is at the same level.
There is a lot of things in AQ to be good for implement for AP, for example baro-gps fusion for keeping altitude, compass using - in the AQ there was NO problem with compass. I think it is worth to try implement it.

MohamedAliRashad, 2019-03-19:
@magicrub 
I can't find any resources for EKF3 except this PR [#5336](https://github.com/ArduPilot/ardupilot/pull/5336)
why no one attempted to add more information with it's own section in Learning_code section in Ardupilot_wiki ?

bys1123, 2019-03-19:
Maybe a good idea

lvale, 2019-03-20:
@priseborough do you have an opinion and guidance ?

priseborough, 2019-03-20:
I've written a prototype UKF version of the EKF3 state vector for a client and came to the conclusion that the performance gain was marginal and the increase in processing requirements was significant preventing it running on a STM32F4. As a result the client client did not pursue the project past the prototype phase.

closed 2019-03-20 by priseborough

bys1123, 2019-03-20:
@priseborough Would UKF be practical on STMH7 or RT1062?

bluuu81, 2019-03-20:
Maybe someone can adopt ready made UKF from AQ ? as olliw42 said it is working for AQ on F4 from years (and issues)