ArduPilot/ardupilot #10782: Matek f405 Mini as Revo Mini alternative.

FPSychotic, 2019-03-11:
Hi just ask if is possible add ChibiOS/APM to Matek 405 Mini.
It Should be a great alternative to Revo Mini as it have pad for ESC telemetry, a very reduced size, a PBD that power it nicely a nd collect the telemetry signal wires. 
I think it has all the features of the big brothers but Im not sure.
For rovers and submarine that need less input/outputs clould be a great board.

Thank you by the effort!!

auturgy, 2019-03-11:
It doesn’t have a baro, so you’d need to add one externally.

Regards,

James

IamPete1, 2019-03-11:
looks like the betaflight target is the same as the standard F405 so it will probably also work on the Arducopter F405 build. Although as James has mentioned no baro and no SD card.

FPSychotic, 2019-03-11:
Rovers, boats, sub-marines dont need baro, and it has internal memory for
logs I think. Even in small robotics/indoor drones and antenna trackers can
be very reliable I think. It is a lot more stackable and safe more space
than revo-mini.
I that reasons still think can be great . For big drones we have other
things better obviously.
Im also thinking in robotics (ROS)

Regards.

IamPete1, 2019-03-11:
if you have one maybe you could give it a try on [this build] (http://firmware.ardupilot.org/Copter/latest/MatekF405-STD/)

FPSychotic, 2019-03-11:
Yes I have one, If somebody guide me I can try, but I never used arducopter, just today a mini pix that got briked all day XD.
There is risk to brick my board? I cant buy other really

IamPete1, 2019-03-11:
should be no risk, you can flash using the betaflight configurator as [outlined here](http://ardupilot.org/copter/docs/common-loading-firmware-onto-chibios-only-boards.html?highlight=flash%20beta)

IamPete1, 2019-03-25:
I will close this pending a update from @FPSychotic

closed 2019-03-25 by IamPete1