ArduPilot/ardupilot #10794: Optitrack position feedback to Rover

telegek, 2019-03-12:
#### Issue details
_Please describe the problem, or desired feature_

Hey guys, we are trying to implement indoor navigation with Optitrack mocap system using ROS. We follow the steps to configure the rover as follows: http://ardupilot.org/copter/docs/common-optitrack.html#configuration-the-drone. Then using **vrpn_client_ros** we are relaying topic to _mavros/mocap/pose_. Seems like EKF2 receives the position, can be seen in the console where **mavros apm** is launched. However, the topic _/mavros/local_position/pose_ stays empty. Also, we tried to feed mocap data to fake gps and vision plugins, but with no success.

Did anyone implement feeding mocap position to arducopter (or rover) from Optitrack using ROS?

#### Platform

[x] Rover

#### Hardware type
_pixhawk_

rmackay9, 2019-03-12:
I don't think anyone has tried the optitrack setup with a Rover but it should work because it is mostly about the EKF which is shared between all the vehicles.  Perhaps @chobitsfan has an idea..

telegek, 2019-03-13:
> I don't think anyone has tried the optitrack setup with a Rover but it should work because it is mostly about the EKF which is shared between all the vehicles. Perhaps @chobitsfan has an idea..

Okay, thank you. Does Optitrack mocap work with copter in ROS environment without NatNet SDK on Windows? I found online that many people face a lot of problems too. Any guide to follow to use motion capture system with the ardupilot? From my experience with PX4, I need to relay to /mavros/vision_pose/pose topic for local position feedback.

chobitsfan, 2019-03-13:
I am sorry but I have little experience with ROS. I think it is possible to decode data streaming from Motive without NatNet SDK. In NatNet SDK readme, it states "It also includes a C/C++ sample showing how to decode Motive UDP packets directly without the use of client libraries". Maybe you can start from that sample.

telegek, 2019-03-13:
I am able to get a PoseStamped message from the Optitrack, which include position and orientation. I am using http://wiki.ros.org/vrpn_client_ros.

The problem is how to feed this data to ardupilot properly and how to configure ardupilot settings. Use fake_gps, vision, mocap plugins?

chobitsfan, 2019-03-14:
fake_gps send HIL_GPS, you need to config ardupilot  GPS_TYPE to 14(MAVLink). Because there is no orientation data in HIL_GPS, you still need a compass and the axis of optitrack must align with magnetic north.
vision plugin send VISION_POSITION_ESTIMATE, mocap plugin send ATT_POS_MOCAP. both are supported by ardupilot. I would recommend mocap plugin, because ATT_POS_MOCAP use quaternion for orientation.

telegek, 2019-03-14:
> fake_gps send HIL_GPS, you need to config ardupilot GPS_TYPE to 14(MAVLink). Because there is no orientation data in HIL_GPS, you still need a compass and the axis of optitrack must align with [...]

Using fake_gps, I get fake gps lat and lon, and can see **/mavros/global_position/local** which is not really the same values coming from the Mocap system, but close and have some offset (0.6). However, topic /mavros/local_position/pose is empty. Is it fine?

chobitsfan, 2019-03-14:
I am not familiar with ROS. Maybe you can check https://discuss.ardupilot.org/t/vision-positionning-experiments-using-skyviper/38348

chobitsfan, 2019-06-18:
hi @telegek 
Thien Nguyen post a very detailed guided about [ROS and Ardupilot integration](https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-1-getting-started-with-the-intel-realsense-t265-on-rasberry-pi-3b/43162/1). Maybe that would be helpful?

telegek, 2019-06-27:
> hi @telegek
> Thien Nguyen post a very detailed guided about [ROS and Ardupilot integration](https://discuss.ardupilot.org/t/integration-of-ardupilot-and-vio-tracking-camera-part-1-getting-started-with-the-intel-realsense-t265-on-rasberry-pi-3b/43162/1). Maybe that would be helpful?

Thank you! Really helpful tutorial.

closed 2019-07-02 by telegek

nirko123, 2019-12-16:
hi @telegek 
did you success to send data position from OptiTrack to ardupilot via ros?

telegek, 2019-12-17:
Hi @nirko123 
No, we didn't. We switched to PX4 and using RC_OVVERRIDE to control the rover.

nirko123, 2019-12-17:
thank you @telegek