ArduPilot/ardupilot #10803: Rover: NAV_SET_YAW_SPEED handling is inconsistent between mission command and command-long

rmackay9, 2019-03-13:
The [NAV_SET_YAW_SPEED](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_SET_YAW_SPEED) command can either be a mission command (executed during Auto mode) or it can be an "immediate" command sent by the GCS or companion computer while the vehicle is in Guided mode.

We handle the two differently which is not good and should be corrected.

param1: **missions** and **immediate** interpret this field as the desired yaw angle (in degrees) but the mavlink spec says centidegrees.  We should adjust the mavlink spec to say degrees

param2:  for **missions** this is interpreted as a target speed in meters/second.  for **immediate commands** this is a ratio of the WP_SPEED value.  I.e. "1" causes the vehicle to drive forward at WP_SPEED, "-1" causes the vehicle to drive backwards at WP_SPEED.  the **mavlink spec** says "normalised speed 0 to 1".  We should change all to interpret this field as meters/second.

param3: **missions** allow this param to be "0" to specify the angle is absolute OR "1" to specify the angle is relative.  **immediate commands** do not use this field nor does the **mavlink spec** mention this. We should add the 0/1 option to the mavlink spec and implement for immediate commands.

I think there are **very** few users of this command so we should feel free to modify the mavlink spec.  ArduPilot is almost certainly the only system to implement this command even in AP I've only known of one users (at my main sponsor EAMS Lab)

amilcarlucas, 2019-03-14:
I guess the first step is to do a Mavlink upstream PR. The commands now also support units="m/s" attributes. Then see how others react to that.

arjun7965, 2019-04-10:
@rmackay9, @amilcarlucas if you could review the PR and give your suggestions, that'd be great!

IamPete1, 2021-08-30:
closed by https://github.com/ArduPilot/ardupilot/pull/12772

closed 2021-08-30 by IamPete1