ArduPilot/ardupilot #10804: Rover: RC and GCS failsafes should not take action if vehicle is disarmed

rmackay9, 2019-03-13:
If the RC failsafe is enabled and the vehicle is powered on before the transmitter the vehicle will enter an RC failsafe leading to the vehicle being switched to RTL mode even though it's still disarmed.

This is slightly annoying because then the user has to remember to switch the vehicle back to an arm-able mode.  Of course users should generally turn on the transmitter before the vehicle but this may not be possible when using receivers that take a while to power-up (i.e Herelink).

So the changes we should probably make are:

- GCS and RC failsafes should not take action if the vehicle is disarmed
- Pre-arm checks should be added to ensure the vehicle is not in a GCS or RC failsafe

Naterater, 2019-03-13:
This is applicable to plane as well. It's annoying because it stresses your servos.

flyfisher2, 2019-03-13:
Agree with Naterater, please add to PLANE.
Although the normal procedure is to switch the tx on first,
there are times when this may not be convenient/desirable.

WickedShell, 2019-03-13:
> Pre-arm checks should be added to ensure the vehicle is not in a GCS or RC failsafe

Absolutely/no questions from me.

> GCS and RC failsafes should not take action if the vehicle is disarmed

This I'm not sold on, as it's actively useful for testing the failsafe behavior/threshold set on the RC transmitter.

Naterater, 2019-03-13:
Then just add an option to THR_FAILSAFE as 3. Enable when armed.

FS_GCS_ENABL is a little more difficult to do as there are already multiple options there. I would be OK with another parameter enabling it when armed.

Overall I think the defaults should change so that failsafe is not enabled during disarmed. Uncommanded mode changes because you don't connect telemetry fast enough is VERY annoying.

peterbarker, 2019-03-14:
On Wed, 13 Mar 2019, Nathan E wrote:

> This is applicable to plane as well. It's annoying because it stresses your servos.

This is a feature - so you can check that your failsafes are actually 
working.

rmackay9, 2019-03-14:
I actually agree with Nate and @flyfisher2 and I suspect we could come up with another way for users to test the failsafes without having servos moving all over the place.  For example in Copter it will still show "Failsafe" on the screen it just won't take the action.  In any case, we are slowly bringing the vehicles together in all sorts of ways so maybe even though we don't have agreement now, perhaps we will someday.

flyfisher2, 2019-03-14:
Coming from Copter I am probably a bit biased but also think that a warning without actually selecting RTL mode is adequate.
Perhaps the warning could include the intended action?

Naterater, 2021-07-26:
This really should happen now with all of the other functionalities built-in.