ArduPilot/ardupilot #10809: Quadplane:

1iggy2, 2019-03-14:
**### Feature request**

_**Is your feature request related to a problem? Please describe.**_
There is not an option for an Octo I (as in India) configuration of motors in the Copter/Quadplane Q_FRAME_TYPE. I see there is an H configuration but due to the geometry that won't be compatible.

_**Describe the solution you'd like**_
I'd like to see the addition of an Octo I class or an ability to rotate the Octo H class. Or an option in MissionPlanner to create custom layouts of motor placements.

_**Describe alternatives you've considered**_
I was looking for a guide to create my own custom frame class (https://github.com/ArduPilot/ardupilot/issues/908) but I did not succeed in finding it. I am not able to alter the Quadplane's geometry.

_**Platform**_
[  ] All
[  ] AntennaTracker
**[X] Copter
[X] Plane**
[ ] Rover
[  ] Submarine

Thank you

auturgy, 2019-03-14:
Octo “I” will be dynamically the same as Octo “X”. Please try that.

closed 2019-03-14 by auturgy

1iggy2, 2019-03-14:
> Octo “I” will be dynamically the same as Octo “X”. Please try that.

That was my assumption, our project team had a safety check with an adviser and they said for a craft of our size (10lbs, 32 inch width, 20.5 inch length) that is not a good assumption. I should have included that information originally. Does that change anything?

auturgy, 2019-03-14:
Doesn’t change a thing.

Regards,

James

1iggy2, 2019-03-16:
> Doesn’t change a thing. Regards, James
> […](#)

Hello,
I was going through the process of writing my own configuration. Our safety adviser is concerned that the craft will not be able to balance the moments if the configuration is not set for our layout. I edited it here (https://github.com/1iggy2/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMatrix.cpp) lines 657-666. Is that the only edit I needed to make to include the new frame type? Also how do I compile the library into a file I can flash onto the Holybro Pixhawk 4 and how do I flash it onto the board. Currently my plan is to use QGroundControl to flash custom firmware. What file type does it need to be and how do I compile the code into that file type?

Sorry to be a bother about this,
Cameron

auturgy, 2019-03-16:
I can argue with your supervisor another time :)
The most succinct way to answer your questions is https://discuss.ardupilot.org/t/custom-motor-mixing-calculator/38035
And have a look at the related videos Chris uploaded to his YouTube (iforce2d).

Regards,

James

rmackay9, 2019-03-17:
Is it really the same as an X though?  I suspect an X will work fine but it does look like an H that's been rotated 90 degrees.  I suspect you may be able to swap the roll and pitch factors of an H to get what you're looking for.

auturgy, 2019-03-18:
The only difference between X and H is yaw response (frame twist countering the desired response), so motor directions are reversed.
Geometric asymmetry is dealt with by separating roll and pitch gains. Hence a H rotated 90deg is dynamically a stretched X: an X with different pitch and roll gains.
What we probably should clarify is that ArduPilot H is actually a sideways H (motor arms lateral not longitudinal), so “capital I” is probably ArduPilot H.
http://ardupilot.org/copter/docs/frame-type-configuration.html

Regards,

James

rmackay9, 2019-03-18:
Wow, I have never seen that wiki page (at least not recently).  We need to get rid of those motor order diagrams and instead refer people to [this page](http://ardupilot.org/copter/docs/connect-escs-and-motors.html).

The X and H are the same for a quad but not for an OctaCopter..

1iggy2, 2019-03-18:
> Wow, I have never seen that wiki page (at least not recently). We need to get rid of those motor order diagrams and instead refer people to [this page](http://ardupilot.org/copter/docs/connect-escs-and-motors.html).
> 
> The X and H are the same (except for yaw) for a quad but not for an OctoCopter..

So rmackay9 you are claiming that the assumption of X being the same as H is not a good one. If you see where I forked the code [here](https://github.com/1iggy2/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMatrix.cpp) you can see how I added the functionality I desire. If I build and flash that code to our controller will Mission Planner's full parameter list show an option for that configuration? If not I will replace the H configuration with our desired code. My team is working on figuring out how to build the code. I've noticed the ArduPilot wiki link for [building the code](http://ardupilot.org/dev/docs/building-setup-windows-cygwin.html#building-setup-windows-cygwin) is sort of lacking. Can you help us with compiling the code by either explaining it or pointing us to a good resource? We will have to edit the code in some way to turn off two of the vertical motors in the beginning of the transition from vertical to horizontal flight due to some aerodynamic interferences. Would you point me to the location of the transition code also, if you know it. I plan on digging but if you happen to know you can save me some solid time.

Sorry for all the requests,
Cameron