ArduPilot/ardupilot #10811: Vector normalize{d}() method dangerous, should have builtin zero check

jo-m, 2019-03-14:
### Bug report
**Issue details**
This is more of an impulse for discussion:

The Vector template class defines the `void normalize()` and `T normalized() const` methods.
As they divide by the vector length, they can cause a division by zero fault.

Even though I did not find a place in the codebase where those are being used without a zero check before, I see some issues:

1. The zero checks in the code base are made inconsistently. Some places use `is_zero()`, some use `fabsf(vec.x) > 0 && ...`, some are right next to where `normalized()` is called, some are a couple lines away.
2. `normalize()` does not immediately make you think "uh gotta look out for zero division here" (seeing `123 / x` in the code is more obvious). Together with above point 1., this creates the danger that someone some day adds or changes an `normalize{d}()` which remains non zero checked.

So, would it be reasonable to add a `is_zero()` check in the `normalize{d}()` methods themselves, and return a value such as `VectorN{0, 0, ..}`?

**Version**
All

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Irrelevant

**Logs**
No logs

khancyr, 2019-03-14:
Hello, 
Indeed, normalise() for matrix and vector isn't protected , the one for quaternion is ! 
Could you open a PR to correct them ? We could see after to simply some logic if the check were done before the normalise() call.
That is sure that fabsf(vec.x) > 0 isn't right but on some place we explicitly don't want to use is_zero as epsilon is to small

peterbarker, 2019-03-14:
On Thu, 14 Mar 2019, Jo M wrote:

> So, would it be reasonable to add a is_zero() check in the normalize{d}() methods themselves, and return a value such as VectorN{0, 0, ..}?

That's possibly worse than the disease :-)

One thing we can do right now is to add a check in SITL that we're not 
dividing by zero - we can panic in SITL all we like :-)

kd0aij, 2019-03-18:
The error here is the attempt to normalize a zero-length vector. That condition must be detected and handled by the caller; the normalize method has no way to know why it's being given something that can't be normalized.

closed 2019-04-04 by jo-m

kd0aij, 2019-04-04:
I think panicking in SITL would be a good idea. But isn't there no chance that attempting to normalize a zero-length vector would in itself cause a hang?

peterbarker, 2019-04-05:
On Thu, 4 Apr 2019, Mark Whitehorn wrote:

> I think panicking in SITL would be a good idea. But isn't there no chance

We now have AP::internal_error().error(...) - this panics in SITL and 
starts to log a failure flag on hardware.