ArduPilot/ardupilot #10812: Plane: large alt.error can be introduced inflight if arming is forced with sub-par GPS fix

vierfuffzig, 2019-03-14:
#### Issue details
This is an issue occasionally reported in local forums and on discuss, see: https://discuss.ardupilot.org/t/altitude-snapped-from-baro-to-gps-alt-tried-to-crash/39447

when arming_require is forced to 0, once EKF starts using GPS, chances are a critically large alt.error is introduced. this usually happens inflight.

while unchecking arming_require when using ekf obviously isn't recommended, sometimes users do, like for testing in armed state where there's no GPS reception, and then tend to forget to set the param back to 1.

imho it's unfortunate to introduce a critically large altitude offset inflight once GPS data are healthy, while using no GPS alt at all likely wouldn't cause any issues.

**Describe the solution you'd like**
prevent switching to GPS alt while is_flying if GPS has never been healthy before

**Describe alternatives you've considered**
completely remove arming_require = 0 option. arming with unhealthy sensors for test- / setup- procedures can be handled via arming checks instead.

Cheers, Basti.

#### Version
plane 3.9.x / 3.10dev

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
fixed wing plane

#### Hardware type
various F4 type boards

#### Logs
https://drive.google.com/open?id=1AbhL_tddVeUCeKU4Gn9s7UU-C3xjUT4H

peterbarker, 2019-04-10:
[image]

I think that might summarise the problem.  `hdem` isn't reset - until the vehicle enters a height-controlled mode again.

vierfuffzig, 2019-04-12:
@peterbarker thanks for looking into this. is this: https://discuss.ardupilot.org/t/ekf-have-a-glitch-in-pos-alt-estimation/41018 related too?
an in-depth assessment is way beyond my scope, however imho it'd be favourable to avoid getting these types of alt errors while flying with two perfectly healthy alt sources.

vierfuffzig, 2021-05-22:
@tridge as you've self-assigned this, do you think you have a chance to pursue this? i'm ok to close this if it doesn't strike a chord in anyone.

hendjoshsr71, 2021-06-21:
I think this is related to the following PR? https://github.com/ArduPilot/ardupilot/pull/17786#

Below the orange shows when the EKF origin first gets set (note the actual Location of the origin doesn't appear in yours logs in the ORGN message.)
[image]

hendjoshsr71, 2021-07-01:
The PR I referenced above got merged and I believe it solves this issue. So I am closing this one. 

If it doesn't please reopen it with a comment.

closed 2021-07-01 by hendjoshsr71

vierfuffzig, 2021-07-04:
thanks @hendjoshsr71 this looks great.