ArduPilot/ardupilot #10820: QGC SurveyScan support via MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET

hamishwillee, 2019-03-15:
QGroundControl allows [StructureScan](https://docs.qgroundcontrol.com/en/PlanView/pattern_structure_scan.html) via "generic MAVLink", but there is a note at the top of the documentation:
[image]

According to https://github.com/mavlink/qgroundcontrol/issues/6824 the issue is that `MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET` has not been implemented.

Clearly it is desirable for ArduPilot if this this feature can be supported.

**Platform**
[ X ] Copter
[ X ] VTOL

rmackay9, 2019-03-15:
This might be a duplicate of this issue [command here](https://github.com/ArduPilot/ardupilot/issues/7658).

hamishwillee, 2019-03-15:
Yes, it is, or will be solved by that issue. I searched for MAV_CMD_DO_SET_ROI_WPNEXT_OFFSET, and survey scan, so sorry, missed that.

closed 2019-03-15 by hamishwillee