ArduPilot/ardupilot #10824: send arming status via frsky telemetry

twistedwings, 2019-03-15:
I would like the arming status to be sent in Tmp1 like INav does.

auturgy, 2019-03-15:
You need to be a bit more explicit.

Regards,

James

twistedwings, 2019-03-15:
I would like the arming status (armed, disarmed, arming prevented) to be transfered via **frsky native** telemetry, in passthru telemetry it is transfered.
One option to transfer the arming status would be Tmp1 like in INav.

Tmp1 in Ardupilot:
Flight mode, sent as an integer: 18 - Manual, 23 - Altitude, 22 - Position, 27 - Mission, 26 - Hold, 28 - Return, 19 - Acro, 24 0 Offboard, 20 - Stabilized, 21 - Rattitude, 25 - Takeoff, 29 - Land, 30 - Follow Me.

Tmp1 in INav:
flight mode, sent as 5 digits. Number is sent as ABCDE detailed below. The numbers are additives (for example: if digit C is 6, it means both position hold and altitude hold are active) :
A : 1 = placeholder so flight mode is always 5 digits long, 4 = failsafe mode
B : 1 = return to home, 2 = waypoint mode, 4 = headfree mode
C : 1 = heading hold, 2 = altitude hold, 4 = position hold
D : 1 = angle mode, 2 = horizon mode, 4 = auto tune mode, 4 = passthru mode
E : 1 = ok to arm, 2 = arming is prevented, 4 = armed

lvale, 2019-03-15:
You have 3 possible telemetry options for FrSky. One of them already sends the arming status. Changing any of the other two would break compatibility with all users.

http://ardupilot.org/copter/docs/common-frsky-repurposed.html

twistedwings, 2019-03-15:
what about defining a new sensor like 0420 (distance) ?

IamPete1, 2021-07-26:
@yaapu This is your area, maybe you can comment/close, Thanks.

yaapu, 2021-07-29:
@IamPete1 as @lvale told @twistedwings we cannot modify how the TMP1 and TMP2 sensors are encoded for it would break compatibility with current setups.
If the need to have extra status bits is a real one we might add another sensor (I would use another instance of the TMP1 or 2) to expose the same status mask used by the passthrough protocol