ArduPilot/ardupilot #10844: Sub: enforce configuration of 'disarm' button

jaxxzer, 2019-03-19:
It should be a requirement that there is always a button configured with the `disarm` function. At least, we should not arm if there is not a disarm button configured.

peterbarker, 2019-03-19:
On Tue, 19 Mar 2019, Jacob Walser wrote:

> It should be a requirement that there is always a button configured with the disarm function. At least, we should not arm if there is not a disarm button
> configured.

You can arm/disarm via mavlink on Sub.  Perhaps adding a check that if you 
have an arm button then you also have a disarm button?

jaxxzer, 2019-03-19:
> You can arm/disarm via mavlink on Sub.

I think this does not preclude the potential to require the disarm button to be configured in any event.

jaylthegreat, 2019-07-08:
I've made a PR solving this issue if you want to assign it in the meantime

Williangalvani, 2021-08-05:
This got addressed in #11749

closed 2021-08-05 by Williangalvani