ArduPilot/ardupilot #10855: Rover: add STICK MIXING

magicrub, 2019-03-20:
Add param [STICK_MIXING ](http://ardupilot.org/plane/docs/parameters.html#stick-mixing-stick-mixing) (or equivalently named) param to do what plane can do; mix in user input while in AUTO mode(s). This is helpful to make manual adjustments in AUTO/GUIDED withotu having to switch in and out of modes.. allow a nudge.

rmackay9, 2019-03-21:
I think this could be OK as long as we completely turn off autonomous control while this is happening.   I'm not a big fan of just adding an additional lateral acceleration to the output from the navigation controllers because I think it will lead to a fight between the autopilot and the pilot... I.e. I-term build up as the auto controller wonders why it's requested turn rate is not being obeyed.  Anyway, happy to discuss though..

mikerob, 2019-03-21:
How does obstacle avoidance under AUTO work?  This would seem to be a similar concept.

Here are my thoughts:
You can either move the target navigation point, or adjust the desired lateral acceleration.

Moving the target point - apply a lateral offset to the L1 target point so that the vehicle navigates to a path parallel to the planned waypoint path.  In a transient sense, it's a swerve to avoid an obstacle but keeps going in largely the same direction.  You wouldn't / couldn't use this mode to actually redirect the vehicle long term. No I-term build up problems because the whole system is just aiming for a new point.

Adjust lateral acceleration - apply an adjustment to the desired lateral acceleration between the navigation controller and the steering controller. This is basically adding a new turn rate on top of the nav controller, which would let you navigate more freely. The only I-term you have to worry about is the cross-track error integrator, which you could ?fairly easily? inhibit during override. 

How the pilot interfaces is another challenge.

Target point: 1) Set a maximum "offset" as a parameter, then stick % maps to the % of that offset applied. When you release the stick, the target point moves back to the planned track and the vehicle re-acquires.  You end up with a pre-defined maximum deviation which can be good or bad.  YTou can't really drive around in circles with this - you're always still basically following the path.

2) Stick input is a lateral velocity of the offset. No limit to how much you can offset the path, but no intuitive return when you release the stick.  Good, however, if you want to maintain a steady offset. Need a function to re-acquire path.

Lateral acceleration: Stick input % maps to 2 x TURN_MAX_G (so you can completely override a max rate commanded turn in the opposite direction). Potentially non-linear. Will automatically return to path when you release the stick.

magicrub, 2019-05-12:
@mikerob sorry for the late reply, I'm just now noticing your post. The idea is to mix stick driver input into the steering controller. Right now it's implemented in this recent merged PR implemented https://github.com/ArduPilot/ardupilot/pull/11244 as a low-level primitive motor mixer but soon it will just nudging the steering desired heading relative to stick input. So, I've now implemented it but soon it will get better.

NickNunno, 2019-05-12:
Also worth noting that rover does not currently have avoidance/dodge capabilities like copter. We can see an object and stop, but not avoid it. 

To implement this as an avoidance move, it would first require completion of a much larger project. (Also being worked on ATM)

IamPete1, 2021-09-12:
closed by https://github.com/ArduPilot/ardupilot/pull/11244

closed 2021-09-12 by IamPete1