ArduPilot/ardupilot #10863: INS_ENABLE_MASK is undocumented and causes bizarre issues when set incorrectly.

akolobov, 2019-03-22:
### Bug report

**Issue details**

ArduPilot parameter INS_ENABLE_MASK isn't mentioned or described in the documentation (http://ardupilot.org/plane/docs/parameters.html). However, setting it to 0 causes a very confusing issue, whereby ArduPilot doesn't initialize any of the IMUs because it doesn't think it needs to, but then fails initialization checks because of this with an error "Check BRD_TYPE: INS: couldn't initialise sensor" (BRD_TYPE has nothing to do with this failure in this case). 

I ended up with INS_ENABLE_MASK = 0 when using a .param file from an older ArduPlane version on versions 3.9.3 and 3.9.6. Since I hadn't even suspected this parameter's existence, getting to the root of the matter took some time.

This parameter needs to be added to the documentation, described, and never allowed to be set to 0 at least in the ArduPlane and Copter flavors.

**Version**
ArduPlane 3.9.3 and 3.9.6

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
All

**Hardware type**
Pixhawk 4 Mini, but it's just applicable to other Pixhawk-family controllers

**Logs**

rmackay9, 2019-03-27:
Thanks for the report.  I've found the issue (a small typo in the parameter description strings) so I'll push a fix to master shortly and close this once that's been done.  After that the parameter description should appear the next time the ground station refetches the parameter descriptions.  If the MP is being used, try Ctrl-F and then push the ParamGen button.  Txs!

akolobov, 2019-03-27:
Thanks for fixing this so quickly!

andyp1per, 2019-11-17:
@rmackay9 this is still not showing up in the Mission Planner docs despite your fix being in for a very long time, I'm wondering whether that's because its the only option under the SUBGROUPINFO entries in AP_InertialSensor.cpp?

andyp1per, 2019-11-17:
I set it to 2, to only enable the second IMU and on Copter 4.0rc2 I get this:

Config error: INS: unable to initialise driver

and HWID still shows both IMUs

peterbarker, 2020-11-25:
Definitely in the docs now.

This problem doesn't reproduce on master, but does on 4.0.x.  I don't think we care enough to backport any fix to 4.0.x, but feel free to prove me wrong :-)

closed 2020-11-25 by peterbarker

kayodedml, 2022-08-17:
help i cant find it under parameters list