ArduPilot/ardupilot #10874: Chibios: random parameter resets to default

vierfuffzig, 2019-03-24:
There seems to be an unresolved issue with random parameter resets in master. reports and respective communication is scattered over gitter-channel, discuss and local forums, so i'm opening this with the intention to collect information in a single place.

reference: https://discuss.ardupilot.org/t/hard-landing-causes-all-parameters-to-be-reset-to-defaults/37189

afaik reports cumulate with recent master firmwares plane / copter, dating back to somewhere around mid january. i myself have never met any reset issues using plane master prior to that date, but some reports do seem to date back further.

resets are random and occasional, i haven't yet found a way to provoke them reliably enough to allow a bisection.

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
various F4-type boards

**Logs**
none available

tridge, 2019-03-24:
thanks! good idea to gather here

vierfuffzig, 2019-03-24:
just ran into a reset with a revomini on today's plane master. even had console running during boot, but it don't show anything beyond correct boot prompts.

here's respective MP logs:

https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/reset_issue_MPlog.rlog
https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/reset_issue_MPlog.tlog

basti

Hwurzburg, 2019-03-24:
I have sporadic full param resets using Arduplane with F35 clone using latish master. It also occasionally locks in bootloader although it has no UARTs powered other than by FC BEC/USB itself. No direct correlation to param resets. No flash memory or SD capability. No issues with any other plane/FC several months older firmware.

vierfuffzig, 2019-03-25:
thanks to @sh83 advice i could catch a reset situation with HAL::panic and acquire a memory dump:

https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/crashdump.s19
https://github.com/vierfuffzig/Chibios-miscellaneous/blob/master/crashdumpbin.bin

hope this helps to track down the issue.

sh83, 2019-03-26:
Based on the dump @vierfuffzig provided now full sequence of events leaded to the storage reset could be restored.
1. AP_FlashStorage::write_all called with non-zero argument.
2. AP_FlashStorage::write_all calls AP_FlashStorage::write with ofs+max_write out of boundary.
3. AP_FlashStorage::write do not check input args and write invalid data block to the flash
4. Fc is rebooted
5. Invalid data is detected by AP_FlashStorage::load_sector and all storage is resetted.

All this caused by https://github.com/ArduPilot/ardupilot/commit/91e5f56 commit.
My suggestions:
1. Revert commit https://github.com/ArduPilot/ardupilot/commit/91e5f56, as it do not do advertised thing
2. Add check to the write method to do not write invalid data

1wick, 2019-03-27:
I noticed the date on that commit is Jan20.. This has been happening to mine since November.  
The first day I installed the version of master w/ flash support it happened... And about twice a week ever since.

sh83, 2019-03-27:
@1wick Probably there is another problem - not the case the @vierfuffzig found. For example https://github.com/ArduPilot/ardupilot/issues/7670 is still not fixed.
The way to debug this is to make the same dump as above.

1wick, 2019-03-27:
Hi, I deleted my previous post before I noticed you replied..  
I looked at the forum and my first post about the issue is actually in January, so maybe I'm wrong about it being November.   (Holiday season being busy and all...)

Does anyone remember when flash support was first introduced for chibios?  I don't remember the date, but I am sure it happened the day I loaded a copy w/ flash support.

sh83, 2019-03-27:
@1wick . It was the same day! 21 jan, https://github.com/ArduPilot/ardupilot/commit/e3c9f10e912e87f18827626ef0bd99d867dd79de
So, all versions with dataflash logging have this bug

magicrub, 2019-03-27:
pending fix: https://github.com/ArduPilot/ardupilot/pull/10927

vierfuffzig, 2019-03-29:
thanks everyone, closing.

closed 2019-03-30 by vierfuffzig