ArduPilot/ardupilot #10879: Copter: SmartRTL may yaw 180° in final stage of landing

AndKe, 2019-03-24:
### Bug report

SmartRTL may suddenly decide to do a nose-in landing, it will just yaw around somewhere close to the ground, for no apparent reason.  
This can be dangerous for any copter that can gain some lateral speed during the yaw (due to bad magnetometer data)

[image]

**Version**
AC3.6.7 release

**Airframe type**
Hexa

**Hardware type**
Pixhawk

**Logs**
https://drive.google.com/file/d/18E-fXv75ho37t6h2sGFDLYlVXK2BpjXz/view?usp=sharing

amilcarlucas, 2019-04-23:
Can you re-test with master ? A change went it today that might have fixed it.

AndKe, 2019-04-24:
Hi, tested today's master.
Unable to reproduce (good news)  it flew as expected.  - but the conditions were good, no wind.
SmartRTL seems to always fly nose-ahead, so I need to test this with gusty wind as well.

Another issue, is that SmartRTL disregards the LAND_SPEED and LAND_SPEED_HIGH, - it just goes fearlessly down.

I suggest that:
1: SmartRTL obey LAND_SPEED below LAND_ALT_LOW
2: inhibit all yawing by playback below LAND_ALT_LOW   (pilot may still control yaw if desired)

Reasoning: 
The pilot may take off and yaw/reposition at low altitude, altitude may drift during the flight.
Later, the same maneuver executed by SmartRTL may mean lateral movement and yawing while almost touching the ground.
Close to ground, magnetometers can be affected by things on/in the ground, so SmartRTL will only do a bad situation worse by yawing around, which, with bad compass data, easily will add lateral movement.