ArduPilot/ardupilot #10889: EKF: position innovation alone for ekf failsafe or primary change

chobitsfan, 2019-03-25:
### Feature request

**Is your feature request related to a problem? Please describe.**
fly Copter-3.5.7, it continues to climbing up even in LAND mode (it finally start to descend in 1400m). After inspect the log, the altitude of 1st EKF3 core (AHRS_EKF_TYPE=3 for this flight) is complete wrong. 
[image]
but 2nd EKF3 core is correct
[image]

**Describe the solution you'd like**
XKF3.IPD is very high short after take off, but no action is taken
[image]
XKF4.SH remains very low util the copter fly up to 1400m
Maybe EKF can change primary or go to failsafe when IPD is really high

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
log: https://drive.google.com/open?id=1avbVCBa1-i9SuasvowmT2dACVdvsTiNq

rmackay9, 2019-03-26:
In parallel I've been working a bit with Paul Riseborough on some improved EKF vibration compensation which is in this branch: https://github.com/rmackay9/rmackay9-ardupilot/commits/copter-fake-clip

It's not quite working though so requires some more effort.  I think the general approach will work though.

chobitsfan, 2019-03-26:
thank you @rmackay9 
It is strange that SP (or SH) remains low (<0.5) when IPN, IPE or IPD already go above 100. Below is another example. IPN already go above 200, but SP still below 0.5

[image]

closed 2020-05-29 by chobitsfan