ArduPilot/ardupilot #10902: Provide metadata for AP_Logger ERR messages

peterbarker, 2019-03-26:
### Feature request

**Is your feature request related to a problem? Please describe.**

There's no metadata available for data-flash logged `ERR` messages; these are just a pair of numbers for most ground control stations.

**Describe the solution you'd like**

Some generated XML/json/? document downloadable from our firmware server for the GCS's to use.

**Describe alternatives you've considered**

Continuing as we always have, and having Michael update MissionPlanner...

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

jaxxzer, 2019-03-26:
Whatever mission planner is doing now for 'human readable' [translations](https://github.com/ArduPilot/ardupilot/pull/10882#discussion_r268940544) is being performed by hand?

peterbarker, 2019-03-26:
Well, there's a copy of the enum: https://github.com/diydrones/MissionPlanner/blob/master/ExtLibs/Utilities/DFLog.cs#L157

I can't see any strings in here anywhere, 'though.

peterbarker, 2024-07-31:
We now generate metadata containing this (https://autotest.ardupilot.org/LogMessages/Copter/_

closed 2024-07-31 by peterbarker