ArduPilot/ardupilot #10910: AutoTest inconsistency

patrickelectric, 2019-03-26:
### Bug report

**Issue details**

Local tests of autotest does not match CI
CI results in different issues running several times

**Version**
master

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[ x ] Submarine

ps: I'm just testing sub

**Hardware type**
SITL

**Logs**

```
AUTOTEST: Failing tests:
  SetHome (Test Set Home) (Did not get SYSTEM_TIME) (see /tmp/buildlogs/ArduSub-SetHome.txt)
>>>> FAILED STEP: dive.ArduSub at Tue Mar 26 11:40:14 2019
FAILED 1 tests: ['dive.ArduSub']

>>>> FAILED STEP: dive.ArduSub at Tue Mar 26 11:42:02 2019 (Did not get SYSTEM_TIME)
Traceback (most recent call last):
  File "./Tools/autotest/autotest.py", line 589, in run_tests
    if run_step(step):
  File "./Tools/autotest/autotest.py", line 386, in run_step
    return tester_class_map[step](binary, **fly_opts).autotest()
  File "/tmp/ardupilot/Tools/autotest/common.py", line 2602, in autotest
    ret = self.run_tests(tests)
  File "/tmp/ardupilot/Tools/autotest/common.py", line 2491, in run_tests
    self.init()
  File "/tmp/ardupilot/Tools/autotest/ardusub.py", line 35, in init
    super(AutoTestSub, self).init()
  File "/tmp/ardupilot/Tools/autotest/common.py", line 1868, in init
    self.apply_defaultfile_parameters()
  File "/tmp/ardupilot/Tools/autotest/common.py", line 287, in apply_defaultfile_parameters
    self.reboot_sitl()
  File "/tmp/ardupilot/Tools/autotest/ardusub.py", line 163, in reboot_sitl
    self.initialise_after_reboot_sitl()
  File "/tmp/ardupilot/Tools/autotest/common.py", line 369, in initialise_after_reboot_sitl
    self.set_streamrate(self.sitl_streamrate()+1)
  File "/tmp/ardupilot/Tools/autotest/common.py", line 361, in set_streamrate
    raise NotAchievedException("Did not get SYSTEM_TIME")
common.NotAchievedException: Did not get SYSTEM_TIME
check step:  dive.ArduSub
Renaming logs/00000002.BIN to /tmp/buildlogs/ArduSub-00000002.BIN
Renaming logs/00000001.BIN to /tmp/buildlogs/ArduSub-00000001.BIN
FAILED 1 tests: ['dive.ArduSub']

>>>> PASSED STEP: dive.ArduSub at Tue Mar 26 11:47:08 2019
check step:  dive.ArduSub

>>>> PASSED STEP: dive.ArduSub at Tue Mar 26 11:52:01 2019
check step:  dive.ArduSub

>>>> PASSED STEP: dive.ArduSub at Tue Mar 26 11:56:55 2019
check step:  dive.ArduSub

>>>> PASSED STEP: dive.ArduSub at Tue Mar 26 12:11:31 2019
check step:  dive.ArduSub

>>>> FAILED STEP: dive.ArduSub at Tue Mar 26 12:40:57 2019 (60)
Traceback (most recent call last):
  File "./Tools/autotest/autotest.py", line 589, in run_tests
    if run_step(step):
  File "./Tools/autotest/autotest.py", line 386, in run_step
    return tester_class_map[step](binary, **fly_opts).autotest()
  File "/tmp/ardupilot/Tools/autotest/common.py", line 2602, in autotest
    ret = self.run_tests(tests)
  File "/tmp/ardupilot/Tools/autotest/common.py", line 2511, in run_tests
    self.close()
  File "/tmp/ardupilot/Tools/autotest/common.py", line 377, in close
    self.mavproxy.expect("Unloaded module map")
  File "/tmp/ardupilot/Tools/autotest/pysim/util.py", line 359, in _expect_callback
    raise pexpect.TIMEOUT(timeout)
pexpect.exceptions.TIMEOUT: 60
check step:  dive.ArduSub

>>>> FAILED STEP: dive.ArduSub at Tue Mar 26 12:47:48 2019 (60)                                                                                                                                                     
Traceback (most recent call last):
  File "./Tools/autotest/autotest.py", line 589, in run_tests
    if run_step(step):
  File "./Tools/autotest/autotest.py", line 386, in run_step
    return tester_class_map[step](binary, **fly_opts).autotest()
  File "/tmp/ardupilot/Tools/autotest/common.py", line 2602, in autotest
    ret = self.run_tests(tests)
  File "/tmp/ardupilot/Tools/autotest/common.py", line 2511, in run_tests
    self.close()
  File "/tmp/ardupilot/Tools/autotest/common.py", line 377, in close
    self.mavproxy.expect("Unloaded module map")
  File "/tmp/ardupilot/Tools/autotest/pysim/util.py", line 359, in _expect_callback
    raise pexpect.TIMEOUT(timeout)
pexpect.exceptions.TIMEOUT: 60
check step:  dive.ArduSub

>>>> FAILED STEP: dive.ArduSub at Tue Mar 26 13:27:19 2019 (Did not get SYSTEM_TIME)
Traceback (most recent call last):
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/common.py", line 1777, in run_one_test
    test_function()
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/ardusub.py", line 182, in <lambda>
    lambda: self.dive_mission("sub_mission.txt")),
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/ardusub.py", line 87, in dive_mission
    self.wait_mode('AUTO')
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/common.py", line 1622, in wait_mode
    raise WaitModeTimeout("Did not change mode")
common.WaitModeTimeout: Did not change mode

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./Tools/autotest/autotest.py", line 589, in run_tests
    if run_step(step):
  File "./Tools/autotest/autotest.py", line 386, in run_step
    return tester_class_map[step](binary, **fly_opts).autotest()
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/common.py", line 2602, in autotest
    ret = self.run_tests(tests)
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/common.py", line 2506, in run_tests
    self.run_one_test(name, desc, func)
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/common.py", line 1795, in run_one_test
    self.check_sitl_reset()
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/common.py", line 1742, in check_sitl_reset
    self.reboot_sitl() # that'll learn it
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/ardusub.py", line 171, in reboot_sitl
    self.initialise_after_reboot_sitl()
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/common.py", line 369, in initialise_after_reboot_sitl
    self.set_streamrate(self.sitl_streamrate()+1)
  File "/mnt/HD/git/blue/ardupilot/Tools/autotest/common.py", line 361, in set_streamrate
    raise NotAchievedException("Did not get SYSTEM_TIME")
common.NotAchievedException: Did not get SYSTEM_TIME
check step:  dive.ArduSub

>>>> PASSED STEP: dive.ArduSub at Tue Mar 26 13:25:02 2019
check step:  dive.ArduSub
```

magicrub, 2019-03-26:
ping @peterbarker I think autotest is suffering from fast-than-realtime problems within MAVproxy that is cpu-load dependent. We're running it too fast.  Something has been getting worse: either Travis is slower or MAVproxy is less stable when dealing with excessive traffic.

peterbarker, 2019-03-27:
On Tue, 26 Mar 2019, Patrick José Pereira wrote:

Please try with a --speedup of, say, 5.

IamPete1, 2021-09-12:
@patrickelectric any update?

khancyr, 2021-09-14:
We could close this. It was mostly due to Travis. Now we more in sync with CI.

closed 2021-09-14 by IamPete1