ArduPilot/ardupilot #10922: nsh> LSM303D and L3GD20 :driver not running

RQRobert, 2019-03-27:
### Bug report
Vibration Dampened Board Sensors (lsm303d,l3gd20  ) are not running
**Issue details**
L3GD20 and LSM9DS0 drivers are not running in my Cube.
_Please describe the problem_
I have recently purchased a new pixhawk2.1.
I power the Cube from the power brick mini and connect it to MP via usb.
I load FW Arducopter FW3.5.7.
In the Full Parameters List , i see the parameters INS_ACC2_ID INS_ACC3_ID have values ZERO.
So i thought the Second IMU is missing .
In order to test the sensors i re-flash the Cube with ArduCopter3.4.6.
In MP>Terminal>nsh> i run command "lsm303d start". I get the following Errors:
LSM303D probe failed
LSM303D: SPI init failed
LSM303D : Driver start failed
and Running the command : nsh>l3gd20 start ". I get the following Errors:
L3GD20 probe failed
L3GD20 : Driver start failed

**Version**
_What version was the issue encountered with_
I have changed different Arducopter firmwares. 
Change different Vehicle Types but with the same Problem.

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
N/A
**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Cube
**Logs**
_Please provide a link to any relevant logs that show the issue_

[tlog.zip](https://github.com/ArduPilot/ardupilot/files/3011904/tlog.zip)

RQRobert, 2019-03-27:
I want to test all the sensors of Cube. 
Kindly help is there any command to run ardupilot from nsh command terminal.
I have found someone run the code in the below link.
https://github.com/ArduPilot/ardupilot/issues/7069

auturgy, 2019-03-27:
Please use discuss.ardupilot.org for user support.

closed 2019-03-27 by auturgy