ArduPilot/ardupilot #10930: Copter: deprecate SPORT mode

rmackay9, 2019-03-28:
Sport mode is a very rarely used mode so I think we should deprecate it to help simplify Copter (a little bit).  Sport mode is a mixture of Acro and AltHold in that the user controls the lean rates (instead of angles) and climb rate.

It was added by Leonard Hall and I thinking that it might be good for filming but it wasn't really based on a user request.

liang-tang, 2019-03-28:
agree

muramura, 2019-03-31:
With the support of OMNIBUS and KAKUTE, I think that there are people who put ArduPilot on drone racing drone.
This drone racing drone camera is equipped with a fixed angle camera.
Recently, it seems that a promotional video has been taken using a quadcopter called nano drone.
This nano drone camera too. It is fixed at a certain angle.
This use has increased since last year, and this mode is considered to be useful from now on.

rmackay9, 2019-03-31:
@muramura, nice video.  I don't think that most drone racers use Sport mode though.  They will tend to use acro I think.  I think it will be unlikely that we will find a user.

P.S. I've removed @lthall as the assignee because this is something i can take care of.  It's just a change to a definition.

muramura, 2019-03-31:
@rmackay9 san...
I am an acquaintance of me who is shooting like a movie.
It seems that there is a nano drone using OMINIBUS and KAKUTE.
I think it is a good mode to shoot such videos.
ex:
https://youtu.be/HMHC4Pakf0Y

rmackay9, 2019-03-31:
@muramura, if you can ask him to try it and report back that would be great.  I suspect he won't like it and will prefer to stay with acro, stabilize or one of the other modes.

muramura, 2019-04-01:
@rmackay9 san.. Thanks Comments.

Naterater, 2019-04-01:
If I remember correctly, sport mode works very well for steady cinematic shots for camera drones. I'm not sure removing it is the best, but I'm just one vote here, and I don't use it myself.

lvale, 2019-04-02:
Although I don't use it frequently with large drones, have used several times with fixed cameras (too big to the gimbal).

proficnc, 2019-04-02:
I use sport mode! It’s awesome!

magicrub, 2019-04-02:
I see no positive, only huge negatives, for removing this.

rmackay9, 2019-04-02:
Alright, seems like a few people want to keep it so we will leave it alone.

closed 2019-04-02 by rmackay9