ArduPilot/ardupilot #10965: BLHeli passthrough unavailable when THR_MIN <0

Naterater, 2019-04-01:
### Bug report

**Issue details**

Just as title shows, BLHeli passthrough is unavailable when THR_MIN is < 0. The flight controller "roboot" is unavailable according to BLHeli Suite.

**Version**
Master as of 3/22/2019, and I have noticed this since DSHOT reverse thrust has been supported (BLH_REMASK parameter was implemented).

**Platform**
[  ] All
[  ] AntennaTracker
[?] Copter
[X] Plane
[?] Rover
[?] Submarine

**Airframe type**
Dual motor electric plane

**Hardware type**
Cube black with mini carrier board.

**Logs**
N/A - configuration issue

Naterater, 2020-09-25:
@IamPete1, wasn't this addressed? Unfortunately 4.0.6 still has this behavior.

IamPete1, 2020-09-25:
Should be fixed in master, hasn't made it to stable tho

closed 2021-07-26 by IamPete1