ArduPilot/ardupilot #10967: An invalid parameter was found

2357531789, 2019-04-02:
When I was adjusting the yaw parameters of my multi-rotor UAV, when I tried to adjust the parameters MOT_YAW_HEADROOM, I found that changing it had no effect.
My configuration is: 580mm wheelbase, DJI e800 motor and 15 inch propeller, which always show weakness on yaw axis.
Yaw's PID: p0.6, i0.4, d0.08.
Log found that the restriction guidance of MOT_YAW_HEADROOM should have led to higher performance, but when I adjusted it, it was not found to be effective.

2357531789, 2019-04-02:
In addition, I can't confirm whether my basic PID or other reasons have prevented me from touching the MOT_YAW_HEADROOM restriction or other reasons. I just want my UAV to be more powerful, so it feels good.
In addition, for roll and pitch I have been able to set very high values to ensure strong anti-jamming power, but here in Yaw stuck, which is very helpless.

magicrub, 2019-04-02:
@2357531789 have you tried an autotune yet?

2357531789, 2019-04-02:
@magicrub  I've tried it, but I don't really like automatic tuning. It has had accidents, and I feel that my understanding of PID is deep enough, so I don't usually use automatic tuning.

rmackay9, 2019-04-02:
This really feels like a support issue so it should probably go in the forums instead of being raised here in the issues list which is normally reserved for enhancement requests and verified bugs.

I'm pretty sure the yaw-headroom parameter is working as expected it's just that it's difficult to see the effect unless the motor library needs to make a priority decision between roll/pitch, throttle and yaw.

I think we should close this issue.

closed 2021-07-26 by IamPete1