ArduPilot/ardupilot #10971: Reproducable flip on takeoff

h4writer, 2019-04-02:
### Bug report

**Issue details**

I'm using Ardupilot 3.5.7 on my drones. And I've noticed that sometimes my drones would flip on takeoff. There was a long time I couldn't reproduce it, but now I have found a way to reliably get the drone to flip. The drone is a regular 450mm quad frame, using here+ gps with rtk. The motors are T-motor. 

Prerequisted:
- loiter mode
- sticky ground (where the drone won't slide on). (grass is enough for me to get this happening)

Steps to reproduce:
1. arm the drone
2. give a little bit throttle (i.e. 51%) for a little while to make sure the drone doesn't disarm
3. wait till enough momentum has build up for the drone to start lifting itself
4. the drone will start to move a little bit, because it has not enough momentum to go fully up
5. the drone cannot move left or right due to the sticky ground
6. More and more momentum builds up and the drone will lift partially in the air on 2 of the 4 legs.
7. Instead of leveling out, the props (already in the air) will start to increase throttle little by little
8. After some time the drone will go too much on one side and eventually flip.

**Version**
Ardupilot 3.5.7

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Cube 2.1

**Logs**
(in the logs I used a kill switch to stop it from actually flipping. But if I didn't use the kill switch the drone would flip at that point. Three examples in the log).
[2019-03-28 14-47-14.log](https://github.com/ArduPilot/ardupilot/files/3034414/2019-03-28.14-47-14.log)

rmackay9, 2019-04-02:
I wonder if there is any more we can do about this though. This is basically I-term build up as the vehicle tries harder and harder to move horizontally (because the position according to the GPS always drifts a bit).

I think the wiki advice for new pilots is to take off assertively.  We also have a PILOT_TKOFF_ALT parameter which forces a takeoff to a configurable altitude.

Reducing the maximum lean angle of the vehicle may also help because the current default of 45 degrees may allow the vehicle to lean further than it should.

muramura, 2019-04-02:
@rmackay9 san..
Is it possible to reduce the movement due to the positioning error of GNSS by using OF?
If you can reduce the movement, I would recommend the use of OF.

rmackay9, 2019-04-03:
@muramura, that's a good idea.  I haven't not tested GPS + OF to see if it reduces the drift.. it might!

h4writer, 2020-04-12:
As post-mortem. We found out when it happens.

It seems to happen more often at night and that is because our batteries (18650) are cold and cannot produce enough amps? We didn't see this issue again if we pre-heat our batteries and put heat pads on the batteries.

Now is there something we can do here? Can we detect when we cannot pull enough amps maybe?

IamPete1, 2021-08-15:
Sounds like this was a hardware problem making the takeoff slow enough to suffer from I term windup.

closed 2021-08-15 by IamPete1

h4writer, 2021-08-15:
Just for completeness. The issue is indeed a hardware problem. The motor would go in fail-safe when battery was cold, due to not being able to draw enough power. As a result only 3 out of 4 motors would work and the drone would flip.