ArduPilot/ardupilot #10976: NavEKF2_core::checkGyroCalStatus will take a very long time.

liang-tang, 2019-04-03:
https://github.com/ArduPilot/ardupilot/blob/fe9a9479e6ecdec9c81622cd457d6387834c9e30/libraries/AP_NavEKF2/AP_NavEKF2_Control.cpp#L457
checkGyroCalStatus will take a very long time

### Feature request
Reduce the time
**Platform**
[*] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
_Add any other context or screenshots about the feature request here._

rmackay9, 2019-04-03:
@liang-tang, can "very long time" be clarified?  is the issue that it's slowing down the main loop?

liang-tang, 2019-04-03:
Simulated with Copter 3.6.7, enabled one IMU, other params default, .
There was a very simple test .
result: checkGyroCalStatus  takes 39578 - 5091 = 34487ms
 
[image]
[image]
[image]

rmackay9, 2019-04-03:
I see so it's not actually that the call takes a long time to execute, the issue is that the EKF is taking a long time to learn the Gyro biases after which it then prints the, "checkGyroCalStatus" message?

So the request is perhaps to speed up the time it takes the EKF to estimate the biases after startup.

liang-tang, 2019-04-03:
Yes，absolutely right.

rmackay9, 2019-04-03:
ok great, thanks for updating the title of the issue too.

tommy91, 2019-11-09:
I'm using SITL with Gazebo to simulate a Copter and in my case the startup process stucks in checkGyroCalStatus(), which keeps returning false, for minuter, not seconds.. I've tried both EKF2 and EKF3 but same result.. Are there any updates? Have you found a solution to speedup the process? Thanks

IamPete1, 2021-09-12:
@tommy91 I would just not use the EKF, its bey fussy with simulators if there physics is not quite correct. Set `AHRS_EKF_TYPE` 10

IamPete1, 2021-09-12:
@liang-tang any update? Is this still a issue?