ArduPilot/ardupilot #10977: Octo "I" Configuration

1iggy2, 2019-04-03:
### Feature request
I would like for there to be an Octo "I" as in India motor mount added into Ap_MotorMatrix.cpp.  I have forked the directory and edited the Octo_H configuration, but we can not build with waf. We have successfully built one piece of firmware, but the motors where in the reverse of the correct position.  I am currently able to flash the most recent release through Mission Planner. Also, I've been told that Octo X is dynamically the same as the configuration I proposed, but there is some disagreement in that fact.

**Describe the solution you'd like**
I would like on line 657 for this code to be inserted, along with any additional changes need to make it compatible with Mission Planner.

                case MOTOR_FRAME_TYPE_I:
                    add_motor_raw(AP_MOTORS_MOT_1,  0.333f, -1.0f, AP_MOTORS_MATRIX_YAW_FACTOR_CW,  1);
                    add_motor_raw(AP_MOTORS_MOT_2, -0.333f,  1.0f, AP_MOTORS_MATRIX_YAW_FACTOR_CW,  5);
                    add_motor_raw(AP_MOTORS_MOT_3,    1.0f, -1.0f, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 2);
                    add_motor_raw(AP_MOTORS_MOT_4,  0.333f,  1.0f, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 4);
                    add_motor_raw(AP_MOTORS_MOT_5, -0.333f, -1.0f, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 8);
                    add_motor_raw(AP_MOTORS_MOT_6,   -1.0f,  1.0f, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 6);
                    add_motor_raw(AP_MOTORS_MOT_7,   -1.0f, -1.0f, AP_MOTORS_MATRIX_YAW_FACTOR_CW,  7);
                    add_motor_raw(AP_MOTORS_MOT_8,    1.0f,  1.0f, AP_MOTORS_MATRIX_YAW_FACTOR_CW,  3);
                    success = true;
                    break;

This is the code made by Iforce2d's calculator (https://www.iforce2d.net/mixercalc/) for our specificdesign:

add_motor_raw(AP_MOTORS_MOT_1, -0.517, 0.684, AP_MOTORS_MATRIX_YAW_FACTOR_CW, 1);
add_motor_raw(AP_MOTORS_MOT_2, 0.517, -0.684, AP_MOTORS_MATRIX_YAW_FACTOR_CW, 2);
add_motor_raw(AP_MOTORS_MOT_3, -1, 0.684, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 3);
add_motor_raw(AP_MOTORS_MOT_4, -0.517, -0.684, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 4);
add_motor_raw(AP_MOTORS_MOT_5, 0.517, 0.684, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 5);
add_motor_raw(AP_MOTORS_MOT_6, 1, -0.684, AP_MOTORS_MATRIX_YAW_FACTOR_CCW, 6);
add_motor_raw(AP_MOTORS_MOT_7, 1, 0.684, AP_MOTORS_MATRIX_YAW_FACTOR_CW, 7);
add_motor_raw(AP_MOTORS_MOT_8, -1, -0.684, AP_MOTORS_MATRIX_YAW_FACTOR_CW, 8);

Here is an image of the calculator page:
[image]
Here are the parameter's I entered in the calculator based off our design:
7 5 h
5 1 h
1 3 h
7 6 v
5 2 v
1 4 v
3 8 v
6 2 h
2 4 h
4 8 h
5 1 394
2 4 394
7 5 184
6 2 184
1 3 184
4 8 184
7 6 521
5 2 521
1 4 521
3 8 521

**Describe alternatives you've considered**
Could some build an arduplane.APJ from this github page (https://github.com/1iggy2/ardupilot) it has my most recent version, but now every computer we try to build with gives us a dead package. What I call a dead package is when I build with waf and sucessfully get a .APJ, but when the firmware is uploaded to the Pixhawk4 and it times out after 30 seconds saying no heartbeat was detected. 

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
Old Help Requests:
https://github.com/ArduPilot/ardupilot/issues/10809
https://github.com/ArduPilot/MissionPlanner/issues/2103
https://discuss.ardupilot.org/t/driver-issue-with-waf-built-arduplane/40104

My Specs:
Holybro Pixhawk4
Mission Planner 1.3.64 build 1.3.7032.578
Tested on Windows 10, Windows 7, Ubuntu

IamPete1, 2019-04-03:
looks the same as octo x to me
http://ardupilot.org/copter/_images/motororder-octoquad-x-2d.png

1iggy2, 2019-04-03:
That is octoquad X, is that the one you meant when you said Octo X or did you mean this configuration? 
[image]

IamPete1, 2019-04-03:
sorry, yeah I meant a octo quad

1iggy2, 2019-04-03:
Interesting, I'm guessing what my adviser will say here, but won't that configuration fail because the motors are not in the correct position. It won't be able to tell that firing the 6 motor is more effective than the two motor, (numbers from Octo X, to give offset, in octo quad X they would be 3 and 8).  Also I should mention the scale, do these assumptions hold for a 10 lb (4.5kg) craft, what about a 55lb (25kg)?

IamPete1, 2019-04-03:
ah, i see, i though the offset was just how they were displayed, so what you actually want is a sideways H
[image]

1iggy2, 2019-04-03:
Precisely, the Octo "I" configuration is a pretty logical solution to Octoplanes, I feel that the configuration should be included in Ardupilot if quadplane support is going to be further developed. Here is an image of our craft. 
[image]
Also it may be useful to refer to this issue https://github.com/ArduPilot/ardupilot/issues/10809.
What do you think my course of action should be?

IamPete1, 2019-04-03:
I fixed it for you, https://github.com/IamPete1/ardupilot/tree/I_copter

and built [copter for pixhawk4 ](https://drive.google.com/file/d/1QVHvRMCBGSuNu_tjNucA2DPjvOsRvg9l/view?usp=sharing) (on master)

but i guess you actually want plane? once you have flight tested we can PR it into master

1iggy2, 2019-04-03:
Thank you so much, yes could you do a pixhawk4 build for the plane? We will likely be doing a hover rig test tonight and I'll get you the results and maybe a video. Here's an image of our hover rig, the sizes at the exact same and we have weights to simulate moments. 
[image]

1iggy2, 2019-04-03:
Will I be able to select the I copter param from mission planner's full param list?

IamPete1, 2019-04-03:
have built [plane](https://drive.google.com/file/d/1gpNeN_1CLn3P8EFtzMMfJX9vRSPAZBlg/view?usp=sharing)

yeah, keep me posted, I have not checked your coefficients, just finished off the code.

yeah the full parameter list, frame class octo and type 15 for I (it wont be in the description)

1iggy2, 2019-04-03:
Ok awesome, would you be able to check my coefficients? I based what I wrote off the Octo H frame type. I was incorrect initially and it caused cascading error the one time we were able to compile the code. Essentially it just spun, we noticed the roll and pitch reactions were wrong so I basically flipped all the signs in the coordinate system for that code. Do you have an easy way to check that? Maybe the calculator would help. I think it is good, but another set of eyes would really help. 
Thank you so much, you've helped me with a huge headache

IamPete1, 2019-04-03:
the proper way to test this would be to build a realflight model but this would a couple hours of work. The calculator is probably correct. You need to make sure you have your motor orders correct, be aware that the motor test will not do the right thing. You now have the emergency stop function in plane, set RCx_OPTION to 31. That will kill it before anything too bad can happen, test in Qstabalise first.

IamPete1, 2019-04-03:
oh, and looking at you picture i would guess you want the 'quad bits' further forward. so the center of thrust of the motors coincides with the center of gravity of the plane.

1iggy2, 2019-04-03:
Which block of code did you implement into your tree? The one that uses f
or the one with exact values. The one with f is not based off the
calculator, but off reverse engineering the Octo H config and a little bit
of trial and error. We can test how it reacts without props with tilting
our model before we fly. I did notice flipping the mode to manual through
telemetry successfully killed the craft as well.

1iggy2, 2019-04-03:
Will ardupilot not be able to overcome the CG being off by ~10%?

1iggy2, 2019-04-03:
Additionally, we have the thrust to hover with almost 4 motors. We figured
the excess thrust will be able to compensate

IamPete1, 2019-04-03:
you can see the changes I made here: https://github.com/ArduPilot/ardupilot/compare/master...IamPete1:I_copter

I just used your table from above. 

Ardupilot would be probably able to overcome it. But would you rather try and tune it to fly the dart backwards or forwards? Plane's are much more sensitive to cg location a general rule of thumb is the 1/4 chord of the wing. Your better to have you cg there and overcome it in 'quad' mode. Will work just wont be very efficient

1iggy2, 2019-04-03:
> you can see the changes I made here: master...IamPete1:I_copter

Ok, I think we are on the same page. Thank you for the changelog. The CG is
in the ideal location for the plane and we are planning on the octocopter
mode to overcome it. Octocopter are definitely much less sensitive to CG
then planes. We will just force it to overcome in octocopter mode.

1iggy2, 2019-04-04:
@IamPete1 your arduplane.APJ did successfully flash to the holybro Pixhawk4 and allow for mission planner connection. However we were not able to fly tonight due to restrictions in our test space. 

Would you mind sharing your steps for how you build the code? I follow this [documentation](http://ardupilot.org/dev/docs/building-setup-windows10.html#building-setup-windows10) and this [documentation](https://github.com/ArduPilot/ardupilot/blob/master/BUILD.md) using Ubuntu in Windows from the Microsoft Store and it only builds dead firmware. I was also recommended to ask if you could send me a script of the commands you used to build. 

_Also the documentation does not seem to be accurate:_
**In step 4** it tells me to run the script [./Tools/environment_install/install-prereqs-ubuntu.sh] but the environment_install directory doesn't exist and this was the command I substituted [./Tools/scripts/install-prereqs-ubuntu.sh] 
**In step 5** the line [export PATH=/opt/gcc-arm-none-eabi-4_9-2015q3/bin:$PATH] did not exist and I did not remove anything. 

My adviser asked if **Pixhawk4** was the correct board type selection in waf, or if it should be **Pixhawk4Pro**.

The waf commands used to build were:
./waf configure --board Pixhawk4
./waf plane

Sorry for the essay, but you are the first person to get me onto a new step and I **really** appreciate it.

auturgy, 2019-04-04:
Pixhawk4Pro (H7) boards aren’t available yet (still in testing), so Pixhawk4 (F7) would be correct.

1iggy2, 2019-04-04:
Whoops

IamPete1, 2019-04-04:
> ./waf configure --board Pixhawk4
> ./waf plane

this is exactly what I did, should work, I think if you get a .apj at all its probably compiled OK. I don't know of anything that would cause you to get a broken .apj except code bugs that get past the compiler.  Maybe you could clone my know good branch and try compiling that.

1iggy2, 2019-04-04:
It fails when I try to compile the master Ardupilot branch as well as my
fork where. I test changes. Are you saying that the master branch may not
be good to compile? If that's true, where should I fork from for changes?
What are your computer specs and how did you set up the build environment?

IamPete1, 2019-04-04:
you branch wont compile I think. Try master first, then my branch. The place to ask for help is gitter
https://gitter.im/ArduPilot/ardupilot

1iggy2, 2019-04-04:
The master is what failed to compile when I tried last night. My branch also fails. I will test your branch later today. Thanks for that link, I hadn't heard of that before.

1iggy2, 2019-04-04:
[Building-Peter_s-Code-Step-by-Step.pdf](https://github.com/ArduPilot/ardupilot/files/3044853/Building-Peter_s-Code-Step-by-Step.pdf)
Here is the test of building Peter's branch.

1iggy2, 2019-04-23:
Final Update:
I was able to successfully build the code after partitioning my computer and running Ubuntu 16.04 natively. The documentation was sufficient. I branched ardupilot and created my own frame class, see [https://github.com/1iggy2/ardupilot] although it may be changed in the future. I successfully test flew it today and there was a minor issue that we only had minimal yaw control. I may change the orientation and then test again. I will create a pull request soon and have it added to the master. If you experience a similar issue feel free to send me a message on here or gitter. 
If you're reading this best of luck,
Cameron Gable

closed 2019-04-23 by 1iggy2

1iggy2, 2019-10-07:
I have also found that WSL defaults to 18.04 when I installed it. I believe this is the main issue with building the code in WSL. Please ensure you are on 16.04.