ArduPilot/ardupilot #10979: SystemTime message, Synchronize problem.

HTaeHyeon, 2019-04-03:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

### Bug report

When I send SystemTime message, It won't synchronize time.

Drone's rtc_type parameter is 1 (Message SystemTime).

_Please describe the problem_

[image]

https://github.com/ArduPilot/ardupilot/blob/820f9bd2606a861027f066919e006b50adea00c1/libraries/AP_RTC/AP_RTC.cpp#L42

I think this part has problem.
That Comment is "check it's from an allowed sources".
So I tried to change below.

if(!(allowed_types == type))
{
       return;
}

It works very good.

I wonder if it is correct.

Check Please.

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[V ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quad

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Pixhawk

**Logs**
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2019-04-03:
I think maybe if you set the RTC_TYPES parameter to "2" (for Mavlink) then it will work.

Perhaps we should add a @Values to the parameter description so that it's easier to understand

HTaeHyeon, 2019-04-04:
but I saw below comment in source.

// @Param: _TYPES
    // @DisplayName: Allowed sources of RTC time
    // @Description: Specifies which sources of UTC time will be accepted
    // @Bitmask: 0:GPS,1:MAVLINK_SYSTEM_TIME,2:HW
    // @User: Advanced
    AP_GROUPINFO("_TYPES",  1, AP_RTC, allowed_types, 1),

Even if I set RTC_TYPES "2", it worked with HW Time(like starting 01. 01. 1970. 00:00:00).

rmackay9, 2019-04-04:
@HTaeHyeon, I think the confusion is coming from @bitmask vs @values.  what you're seeing above is @bitmask.. so those are the bit positions. i.e. 0 = "1", 1 = "2", 3 = "4".

HTaeHyeon, 2019-04-04:
I see. Thank you so much.

I confuse that!! Thank you so much.

closed 2019-04-04 by HTaeHyeon

peterbarker, 2019-04-04:
On Wed, 3 Apr 2019, HTaeHyeon wrote:

> // @param: _TYPES
> // @DisplayName: Allowed sources of RTC time
> // @description: Specifies which sources of UTC time will be accepted
> // @bitmask: 0:GPS,1:MAVLINK_SYSTEM_TIME,2:HW
> // @user: Advanced [...]

Is your GCS sending SYSTEM_TIME messages?